Xinyuan-LilyGO/T-Display-S3 #168: Constantly getting i2c errors when using touch (nothing else connected to i2c bus)

shyney7, 2023-06-29:
Im using the Lilygo T-Display-S3 CAP. Touch version and if I implement the touch functionality to my code I'm constantly getting the following i2c errors:
```
[383725][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[401620][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[403552][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
```
The problem is also that as soon as the errors occur the whole process loop is slowed down. This is unacceptable.
Nothing else is connected to the i2c bus and this seems to be a problem with every unit since all 4 units that I bought have this problem.

I'm not doing anything special I'm just changing a variable with the touch function:
```

#define TOUCH_MODULES_CST_SELF
#include <Arduino.h>
#include "TFT_eSPI.h"
#include "pin_config.h" //default TDisplay-S3 pin occupation
#include "TouchLib.h"

//TFT
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sprite = TFT_eSprite(&tft);
//                                     wire, 18, 17, 0x15, 21
TouchLib touch(Wire, PIN_IIC_SDA, PIN_IIC_SCL, CTS820_SLAVE_ADDRESS, PIN_TOUCH_RES);

int deb = 0; //debounce touch
uint8_t currentUIwindow = 0;
void switchTouch();

void setup() {
  Serial.begin(115200);
  // init TFT and Touch
  pinMode(PIN_POWER_ON, OUTPUT);
  digitalWrite(PIN_POWER_ON, HIGH);
  pinMode(PIN_TOUCH_RES, OUTPUT);
  digitalWrite(PIN_TOUCH_RES, LOW);
  delay(500);
  digitalWrite(PIN_TOUCH_RES, HIGH);
  tft.begin();
  tft.fillScreen(TFT_BLACK);
  tft.setRotation(1);
  touch.setRotation(1);
  sprite.createSprite(320,170);
  sprite.setTextColor(TFT_WHITE,TFT_BLACK);
  sprite.fillSprite(TFT_BLACK);
  sprite.setSwapBytes(true);
  Wire.begin(PIN_IIC_SDA, PIN_IIC_SCL);
}

void loop() {
  switchTouch();
  delay(10);
}

void switchTouch() {

   if (touch.read()) {
    if (deb==0){
      deb=1;
      TP_Point t = touch.getPoint(0);
      if (t.x >= 160) {
        if (currentUIwindow >= 4 || currentUIwindow < 0) {
          currentUIwindow = 0;
        }
        else {
          ++currentUIwindow;
        }
      }
      if (t.x < 160) {
        if (currentUIwindow <= 0) {
          currentUIwindow = 4;
        }
        else {
          --currentUIwindow;
        }
      }
    }
  }
  else {
    deb=0;
  }

}
```

shyney7, 2023-06-29:
> I'v seen this as well
> 
> have a look at lvgl i2c_manager Arduino has no concept of multicore - so it's fly by night or you take care yourself pretty sure one task doing i2c is rescheduled before the i2c transaction is done and another task also does some i2c io

I dont understand why there are scheduling problems if nothing else is connected via i2c? Also I dont want to be forced to do RTOS or concurrency programming with all its pitfalls.

mhaberler, 2023-06-29:
how exactly can you run touch on core 0 and loop on core 1 if you are saying this is single-threaded?

why not post a repo for inspection

that said: FreeRTOS-based Arduino usually has some background tasks like for Wifi event handling
you typically do not see them except if you dig deeper

shyney7, 2023-06-29:
With :
```
TaskHandle_t Task1;
xTaskCreatePinnedToCore(
      Task1code, /* Function to implement the task */
      "Task1", /* Name of the task */
      10000,  /* Stack size in words */
      NULL,  /* Task input parameter */
      0,  /* Priority of the task */
      &Task1,  /* Task handle. */
      0); /* Core where the task should run */

Void Task1code( void * parameter) {
  while(true) {
    if (touch.read()) {...
    (...)
  }
```

lewisxhe, 2023-06-29:
Are you sure you are using the CST820 touch?

shyney7, 2023-06-29:
> Are you sure you are using the CST820 touch?

Yes the touch_test example only works if I use `#define TOUCH_MODULES_CST_SELF `
The touch functionality is also working as intended in my code but I'm constantly getting those i2c errors which also seems to block code execution irregularly for a short time. And also it seems that in irregular time intervals touch is beeing registered although nothing touched the display. Its really annoying and makes the use of the touch functionality unusable.

lewisxhe, 2023-06-29:
Oh! If using CST820, if the touch chip does not detect being pressed for a certain period of time, and ESP32 accesses the touch device, then CST820 will not return any data. This is mainly due to the access mechanism of CST820 touch, which is not a fault

lewisxhe, 2023-06-29:
By the way, the CST820 is not a standard I2C device This is very annoying to me

shyney7, 2023-06-29:
This is the complete code btw: https://github.com/shyney7/TDisplay-S3_PMX_Ground

lewisxhe, 2023-06-29:
You can add fully closed internal debugging printing in ***platformio.ini***
```

build_flags =
    -DCORE_DEBUG_LEVEL=0 

```

shyney7, 2023-06-30:
Disabling debug messages doesn't fix the problem it just hides it.
Also when the random touch registration occurs the values for the x and y coordinates are both `4095` which is completely out of range of the real display size. At least I can now ignore touch values higher than 320 for y and 170 for x in my program. But the annoying i2c errors is what bothers me since the loop itself is being blocked for a short time repeatedly due to these errors.

shyney7, 2023-06-30:
Whats also weird is that if I keep touching the display the touch functionality works completely fine.
But as soon as I dont use touch the i2c errors keep occuring.

lewisxhe, 2023-06-30:
> Oh! If using CST820, if the touch chip does not detect being pressed for a certain period of time, and ESP32 accesses the touch device, then CST820 will not return any data. This is mainly due to the access mechanism of CST820 touch, which is not a fault

I think I have already explained.

mhaberler, 2023-06-30:
> Whats also weird is that if I keep touching the display the touch functionality works completely fine. But as soon as I dont use touch the i2c errors keep occuring.

well touch changes relative timing 
it's likely a race condition

either use a mutex around I2C transactions, or try lvgl i2c_manager

shyney7, 2023-07-04:
> > Whats also weird is that if I keep touching the display the touch functionality works completely fine. But as soon as I dont use touch the i2c errors keep occuring.
> 
> well touch changes relative timing it's likely a race condition
> 
> either use a mutex around I2C transactions, or try lvgl i2c_manager

If nothing else except the touch IC is using the i2c bus why should there be a race condition?
Nevertheless I used a pthread mutex and locked it before calling the touch functionality and unlocked it after that, and I'm still getting the same i2c errors as shown above including touch registrations of 4095 and 3331.

shyney7, 2023-07-07:
This problem was fixed in the espressif esp-bsp touch driver repo 3 weeks ago with this pull request: https://github.com/espressif/esp-bsp/pull/181

This is the issue that was created: https://github.com/espressif/esp-bsp/issues/178

Is it possible for lilygo to update the touchlib with this information?

Arielhh, 2023-09-14:
Try the CST816S library. It works. In case you don't need all the functionality of that library you can modify it and remove the gestures.

closed 2024-05-27 by github-actions[bot]