Xinyuan-LilyGO/T-Display-S3 #181: None of my Qwiic (I2C) sensors is recognized

d3mac123, 2023-08-08:
Hi, I am trying to connect 2 Sparkfun sensors (temperature and Lidar) without success (they run fine on an Arduino Nano). No errors besides the "Cannot connect to <sensor name>". Any ideas?

`/******************************************************************************
Example1_Basic_Temperature_Readings.ino
Example for the TMP102 I2C Temperature Sensor
Alex Wende @ SparkFun Electronics
April 29th 2016
~

This sketch configures the TMP102 temperature sensor and prints the
temperature and alert state (both from the physical pin, as well as by
reading from the configuration register.

Resources:
Wire.h (included with Arduino IDE)
SparkFunTMP102.h

Development environment specifics:
Arduino 1.0+

This code is beerware; if you see me (or any other SparkFun employee) at
the local, and you've found our code helpful, please buy us a round!

Distributed as-is; no warranty is given.   
******************************************************************************/

// Include the SparkFun TMP102 library.
// Click here to get the library: http://librarymanager/All#SparkFun_TMP102

#include <Wire.h> // Used to establied serial communication on the I2C bus
#include <SparkFunTMP102.h> // Used to send and recieve specific information from our sensor

// Connections
// VCC = 3.3V
// GND = GND
// SDA = A4
// SCL = A5
const int ALERT_PIN = A3;

TMP102 sensor0;
// Sensor address can be changed with an external jumper to:
// ADD0 - Address
//  VCC - 0x49
//  SDA - 0x4A
//  SCL - 0x4B

void setup() {
  Serial.begin(115200);
  Wire.begin(); //Join I2C Bus
  
  pinMode(ALERT_PIN,INPUT);  // Declare alertPin as an input
  
  /* The TMP102 uses the default settings with the address 0x48 using Wire.
  
     Optionally, if the address jumpers are modified, or using a different I2C bus,
	 these parameters can be changed here. E.g. sensor0.begin(0x49,Wire1)
	 
	 It will return true on success or false on failure to communicate. */
  if(!sensor0.begin())
  {
    Serial.println("Cannot connect to TMP102.");
    Serial.println("Is the board connected? Is the device ID correct?");
    while(1);
  }
  
  Serial.println("Connected to TMP102!");
  delay(100);

  // Initialize sensor0 settings
  // These settings are saved in the sensor, even if it loses power
  
  // set the number of consecutive faults before triggering alarm.
  // 0-3: 0:1 fault, 1:2 faults, 2:4 faults, 3:6 faults.
  sensor0.setFault(0);  // Trigger alarm immediately
  
  // set the polarity of the Alarm. (0:Active LOW, 1:Active HIGH).
  sensor0.setAlertPolarity(1); // Active HIGH
  
  // set the sensor in Comparator Mode (0) or Interrupt Mode (1).
  sensor0.setAlertMode(0); // Comparator Mode.
  
  // set the Conversion Rate (how quickly the sensor gets a new reading)
  //0-3: 0:0.25Hz, 1:1Hz, 2:4Hz, 3:8Hz
  sensor0.setConversionRate(2);
  
  //set Extended Mode.
  //0:12-bit Temperature(-55C to +128C) 1:13-bit Temperature(-55C to +150C)
  sensor0.setExtendedMode(0);

  //set T_HIGH, the upper limit to trigger the alert on
  sensor0.setHighTempF(85.0);  // set T_HIGH in F
  //sensor0.setHighTempC(29.4); // set T_HIGH in C
  
  //set T_LOW, the lower limit to shut turn off the alert
  sensor0.setLowTempF(84.0);  // set T_LOW in F
  //sensor0.setLowTempC(26.67); // set T_LOW in C
}
 
void loop()
{
  float temperature;
  boolean alertPinState, alertRegisterState;
  
  // Turn sensor on to start temperature measurement.
  // Current consumtion typically ~10uA.
  sensor0.wakeup();

  // read temperature data
  temperature = sensor0.readTempF();
  //temperature = sensor0.readTempC();

  // Check for Alert
  alertPinState = digitalRead(ALERT_PIN); // read the Alert from pin
  alertRegisterState = sensor0.alert();   // read the Alert from register
  
  // Place sensor in sleep mode to save power.
  // Current consumtion typically <0.5uA.
  sensor0.sleep();

  // Print temperature and alarm state
  Serial.print("Temperature: ");
  Serial.print(temperature);
  
  Serial.print("\tAlert Pin: ");
  Serial.print(alertPinState);
  
  Serial.print("\tAlert Register: ");
  Serial.println(alertRegisterState);
  
  delay(1000);  // Wait 1000ms
}`

teastainGit, 2023-08-08:
The old Arduinos had their pins fixed by the "Board Type", but ESP32s have various different and user optional I2C pins, so the original sketch needs pin info for ESP32S3 Dev Module.
If you are using the I2C socket at the bottom of the board the Wire.begin(); needs SDA = 43 and SCL = 44:
Wire.begin(43,44);
-Terry

d3mac123, 2023-08-08:
Not using the sockets but the Qwiic/JST connector (which used to work before). I even thought I had broken the connector because, suddenly, the code could not connect with the sensors anymore. I got another board and used the basic sample (just to guarantee my code was not messing up with anything) but no success.

teastainGit, 2023-08-08:
I did mean the JST connector on the bottom of the board.
Your example showed an empty Wire.begin(); It needs to specify the correct pins.
You say "anymore" did you mean that the TMP102 used to work on your LilyGO T-Display S3 ?

d3mac123, 2023-08-09:
Yes, both sensors (with this basic code) used to work before in the S3.

teastainGit, 2023-08-09:
So, I hooked up a TMP102 to my LilyGO T-Display S3 and it works fine with your code.
Make of that what you will.
-Terry

d3mac123, 2023-08-09:
First of all, thank you very much for trying to help. I think something has changed in my board settings because, as I said before, the sample code used to work fine and now it is not (also, I haven't updated any libraries so, I don't think the issue is related to a bad code). Yes, I am using Arduino IDE 2.x. (sorry for such dumb question) What do you mean by latest board definition?

teastainGit, 2023-08-09:
[image]
2nd one down is the latest ESP32 board definition, and the one I used for my test yesterday.

d3mac123, 2023-08-10:
that's the version I have installed (2.0.11) 🙁

Alex Souza

teastainGit, 2023-08-10:
Did you try with Wire.begin(43,44);
Also will you try wiring them to pins 1, 2 as a test?
Pins 17, 18 (SDA, SCL) are also UART and work for me. (just tried it right now)
You need to set Wire.begin(SDA, SCL); though.

d3mac123, 2023-08-10:
Not sure what the (43,44) mean (sorry, I am just a UX designer playing with the esp32) but it DID THE TRICK! All working now. Thank you so much @teastainGit

teastainGit, 2023-08-10:
You are very welcome, my pleasure!
Wire.begin(43,44); means that you are manually specifying the two comm pins for I2c.
The JST Connector  pins are  SDA43, SCL44.

closed 2023-08-11 by d3mac123