Xinyuan-LilyGO/T-Display-S3 #184: Zip file problems

AlanLL, 2023-08-18:
The file was unzipped and copied to the Arduino libaries folder. I had to do this because importing through libaries add  azip file- it gave an error as invalid libary, so i copied as stated.
How ever when compiling a program, i get the error "loading library from c:\Users\Alan_\Documents\Arduino\libraries\T-Display-S3-main: invalid library: no header files found"
Also tthe libary does not show up in the files/examples.
Your help would be appreciated.

lewisxhe, 2023-08-18:
Just put the T-Display-S3 [lib](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/lib) folder in the specified location, and then open any one of T-Display-S3/example to start the next step

AlanLL, 2023-08-18:
My "C:\Users\Alan_\Documents\Arduino\libraries\T-Display-S3-main" folder now only cotains the lib folder.
I selected the  "C:\Users\Alan_\Documents\Arduino\libraries\T-Display-S3-main\lib\TFT_eSPI\examples\160 x 128\Pong_v3" file and tried to upload it, only to get the error as shown in the  error.txt file

[errors.docx](https://github.com/Xinyuan-LilyGO/T-Display-S3/files/12380824/errors.docx)

lewisxhe, 2023-08-18:
[image]

lewisxhe, 2023-08-18:
[image]

AlanLL, 2023-08-18:
I already have TFT_eSPI version 2.5.0 by Bodmer installed.
I only received my T-Display S3. When first plug in it did a wifi scan and the tried to connect to  a Chineseweb site and failed. I tried to load the getting started file bi Volos Projects and it would upload. I then downloaded the firmware tools and uploadedthe tft test firmware. This was succesful and I now have the clourful logo and then all the test fonts.
But I can't up load a program. I have no trouble with LilyGO T-DisplayS3 AMOLED, so why am I having trouble with T-Display S3??

lewisxhe, 2023-08-19:
If you install TFT through the ArduinoIDE library manager_ ESPI, you need to enable Pinmap for T-Display S3 ,See [User_Setup_Select.h](https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/faff18aa753c1745d7609ab859078cea29d9e2fe/lib/TFT_eSPI/User_Setup_Select.h#L131), otherwise it will be the default Pinmap.
T-Display S3 AMOLED does not rely on TFT_ ESPI, so no settings are required

AlanLL, 2023-08-19:
This is my User_Setyp_Select.h:
#include <User_Setups/Setup206_LilyGo_T_Display_S3.h> 
This is the message i get when i try to upload:
"esptool.py v4.5.1
Serial port COM12
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: Wrong boot mode detected (0x8)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2"

How do i get into download mode??
For this program (Volos Project)
[S3ProgressBar-main.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3/files/12385523/S3ProgressBar-main.zip)

lewisxhe, 2023-08-19:
See README FAQ

AlanLL, 2023-08-19:
Where is :
README FAQ
Please??

lewisxhe, 2023-08-19:
https://github.com/Xinyuan-LilyGO/T-Display-S3#faq

AlanLL, 2023-08-19:
I put the factory firmware.bin back, followed the FAQ
and tried to upload. Result-
esptool.py v4.5.1
Serial port COM12
Connecting...

A serial exception error occurred: Write timeout
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1

lewisxhe, 2023-08-19:
Did you follow the order of the FAQs? I'm referring to putting the ESP into download mode. If all else works, take the shot, including keystrokes, send it to the store where you bought it, and let them handle it.

AlanLL, 2023-08-19:
Yes I did follow the FAQ Hold BOOT,press RST, release RST, release BOOT.
Upload code, and it still fails!
I got the T-Display S3 form LilyGO

lewisxhe, 2023-08-19:
You mentioned above that you also have an AMOLED? Is it normal?

T-Display-S3 Please record the operation video and contact the merchant where you purchased it.

AlanLL, 2023-08-19:
The AMOLED display works just fine.
I have started a conversation with LilyGO as to the resetting the upload. Iam waiting to hear back.

AlanLL, 2023-08-22:
Prviously to reset the upload method was not working because I was holding down the BOOT key and pressing I014 button (I thought this was the RST key) Now I Know the RST button is on the side of the BOOT button.
It looks like I've caused my self all of these problems because I did'nt RTFM!

closed 2024-04-09 by github-actions[bot]