Xinyuan-LilyGO/T-Display-S3 #185: Default Serial communication is dead?

contractorwolf, 2023-09-09:
I have a T-display s3 and something strange is happening.  Its stopped doing serial communication over the default Serial port.  I loaded a basic Serial test script that will flip the backlight on and off (so that I can determine that the board is in fact running code.  But the Serial communication seems off, and I have never had something life this happen and I dont know how to fix it.  Here is my test code:

```

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(38, OUTPUT);
  Serial.begin(9600);
  while (!Serial);  // wait for serial port to connect. Needed for native USB
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(38, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(38, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);   
  Serial.println("flip");              // wait for a second
}

```

I get the backlight flipping on and off, but the serial port in Arduino IDE 2 is showing nothing.  Restarted the Serial monitor, my machine, everything but still nothing.  What am I forgetting?  I dont really think the chip is dead I am just blanking or not seeing something critical?

teastainGit, 2023-09-09:
Couple of things to consider:
"while" is not a good idea in a real-time environment, it may stop there "for ever".
Serial print requires USB CDC on Boot "Enabled"
And to enable the LCD power requires output 15 to be on.
The flashing of the BackLight is a cool  indicator of execution unless the backlight comes on...the chip crashes...then reboots, over and over.
I made a user support GitHub repository here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
Maybe this is a long read but very helpful!
-Terry

contractorwolf, 2023-09-09:
ahh, wtf.  I just flipped the same code over to vscode to see if it was the Arduino IDE that was messing up and in VSCode it works fine and logs the "flip"s to the terminal as expected.  I wasted 2 days because of this and it was the IDE

contractorwolf, 2023-09-09:
yeah man, this isnt my normal code I was just dropping my code down to bare bones just to see why my Serial.println() statements werent showing. thx for jumping on though.

teastainGit, 2023-09-10:
I copied your sketch to my Arduino IDE 2.2.1 and it complied and ran with the serial 'flip' message.
ESP board def 2.0.12.
Make of that what you will.

-Terry

contractorwolf, 2023-09-13:
thx for testing it Terry! Always appreciated

closed 2024-04-08 by github-actions[bot]