Xinyuan-LilyGO/T-Display-S3 #188: Can you provide a picture of the antenna section?

Arielhh, 2023-09-20:
You can see it here: https://www.lilygo.cc/products/t-display-s3

teastainGit, 2023-09-20:
Here is a iPhone pic of my LilyGO T-Display S3 .
[image]
Are both the U.FL Ipex connector...AND...the built-in PCB 2D antenna connected to the ESP32?
It looks like you need to move the 0 Ohm resistor 90 deg,  from where it is shipped, to connect the U.FL Ipex connector.
On my LilyGO T-Display S3 boards there is no continuity from the ESP32 to the U.FL centre pin.
Many of us are asking for some clarity.
-Terry

Arielhh, 2023-09-20:
I tried moving the o ohm resistor 90 degrees but same poor results with
various antennas that I purchased.
Also, I tried on a few lilygo youch units, and the same poor results
persist.

sardonicsloth, 2023-10-15:
I would also like clarification here. Has anyone gotten the external ipex connector to work?

JohnSkillman, 2024-03-29:
Still need info on this issue... Im having the same problem

closed 2024-05-13 by github-actions[bot]