Xinyuan-LilyGO/T-Display-S3 #189: Modifying the RESET button behavior

mottibz, 2023-09-20:
Hello,
I have a few of your T-Display-S3 devices and I developed on them a system, integrated with ESP Rainmaker, to control my smart home.
I want to set this up also at a few friends' houses but there is one issue. 
If somebody pushes the RESET button, the device will reset itself immediately. This is not good as they may do this accidentally. 
My question is if there is a way to disable this button from immediately resetting the device but for example, require the user to push it for at least 10 seconds before it will reset the device?
Thanks a lot for your help!!!
Best Regards,
Motti

teastainGit, 2023-09-20:
The Reset button is hard wired into the ESP32.
Best idea is to make your code start and run after a power failure or Reset.
My HVAC ESP_NOW system does this and I cannot tell if there was a power failure!
-Terry

mottibz, 2023-09-21:
Hi, That's what currently happens (my code starts again and works) but some data that is in memory is lost and all starts from scratch. At least if I could get a callback to save info before resetting or something else. I can't afford to do a save to flash for every  small operation... Is there something possible?

mager33, 2024-02-11:
You can evaluate the reason why the ESP32 was reset, so you can detect that the reset button was pressed. Look here: https://github.com/espressif/arduino-esp32/issues/449

mottibz, 2024-02-12:
Thanks for the suggestion! I will use it.
Only problem is that knowing the reason after restart does not help with in-memory data that was lost :-)

closed 2024-04-08 by github-actions[bot]