Xinyuan-LilyGO/T-Display-S3 #191: BURN BOOTLOADER

octopuss83, 2023-09-24:
Hi ,

    Sorry for my english I'm french . I have recently burn my esp32-S3 on a T Display board ( Hight voltage 20V accidently put on GPIO Input My ESP32-S3 is HS . I buy a new ESP32-S3 replace on T display bord and now I want to burn a new bootloader on this chip to restor my ESP32-S3 TDisplay BOARD is it possible ? with usb port ? 
   Thanks for response

teastainGit, 2023-09-24:
"Sorry for my english I'm french . 
I have recently burn my esp32-S3 on a T Display board ( Hight voltage 20V accidently put on GPIO Input My ESP32-S3 is HS 
I buy a new ESP32-S3 replace on T display bord and now I want to burn a new bootloader on this chip to restor my ESP32-S3 TDisplay BOARD is it possible ? with usb port ? "
Hi! I am fascinated that you were able to purchase an identical ESP32 S3R8!
And solder it to your board? Good job!
You do not need a new bootloader because the ESP32 S3 has the bootloader  built-in, in ROM, so you should be able just continue uploading your sketches as before via the USB.
-Terry

octopuss83, 2023-09-25:
OK thanks Terry , I prefer to repair rather than throw away but passionate about electronics I am equipped to repair CMS so at $4 why not. But from your response I just realized that I ordered the wrong version of the ESP32. I ordered the ESP32-S3 and not the ESP32-S3R8. I'm going to see how to adapt that to my card. if it is possible the PB being that I do not have a USB port.

[image]

Thanks for your help

closed 2024-04-07 by github-actions[bot]