Xinyuan-LilyGO/T-Display-S3 #192: Merge changes to TFT_eSPI upstream

alranel, 2023-09-25:
Hello!
At the moment, a modified copy of the TFT_eSPI library is shipped in this repository that users have to manually install in their libraries folder replacing the original TFT_eSPI library. As of today, the upstream library does not seem to work with the examples provided here for T-Display-S3.

This creates multiple issues:

* the library can't be installed/upgraded using the Library Manager in the Arduino IDE
* this copy of the library doesn't get the updates to the upstream library, or gets them with a delay
* it is nearly impossible to use for anyone who needs to have the upstream library installed for other projects

Could you consider making a pull request to the original library with your changes so that can just install the original TFT_eSPI library from the Arduino IDE instead of having to replace it with this modified copy?

As an alternative, if you don't think a pull request is feasible, I strongly recommend to consider moving your library to its own repository and call it with an alternative name like "TFT_eSPI_LilyGo" or "TFT_eSPI_TDisplayS3" so that it can be installed from the Arduino IDE (much better experience for your customers!) and does not collide with the original one.

Thank you :)))

lewisxhe, 2023-10-12:
The upstream TFT_eSPI has added the configuration file of T-Display-S3 and can be upgraded directly. However, after the upgrade, the configuration of T-Display-S3 needs to be re-enabled.

lyusupov, 2023-10-12:
> > consider making a pull request ...

> The upstream TFT_eSPI has added the configuration file of T-Display-S3 ...

https://github.com/Bodmer/TFT_eSPI/pull/2317

closed 2024-04-07 by github-actions[bot]