Xinyuan-LilyGO/T-Display-S3 #195: Esp32-s3 sudden compile error

Davidhndrsn70, 2023-10-11:
I have been working on a project using this board : "[Amazon.com](https://www.amazon.com/dp/B0BF542H39?ref=ppx_yo2ov_dt_b_product_details&th=1)" and using TFT_eSPI. Everything has been working fine until i tryed to add wifi. I started using the following includes:
#include <WiFi.h>
#include <WebServer.h>
#include <WebSocketsServer.h>
#include <ArduinoJson.h>

This lead to an error that lead to me uninstalling Arduino, deleting the Arduino 15 folder (C:\Users\david\AppData\Local\Arduino15), and reinstalling Arduino.

This solved the WIFI issue but now TFT is not compiling. I tried going back through all of the setup steps found in this document for arduino: [GitHub - Xinyuan-LilyGO/T-Display-S3 1](https://github.com/Xinyuan-LilyGO/T-Display-S3)

I went back to using an example project to see if there was somthing wrong with my code however the example gives the exact same error message wihle trying to compile.

This is the code:

#include "TFT_eSPI.h"
#include "install.h"

TFT_eSPI tft= TFT_eSPI();
TFT_eSprite sprite = TFT_eSprite(&tft);

void setup() {
  tft.init();
  tft.setRotation(1);
  tft.setSwapBytes(true);
  tft.fillScreen(TFT_WHITE);
  tft.pushImage(106, 0, 214, 170, install);

  sprite.createSprite(146,170);
 
  sprite.setTextColor(TFT_BLACK,TFT_WHITE);
  sprite.setTextDatum(4);
}

//progress bar variables
int progress=0;
int w=120;
int h=18;
int x=12;
int y=90;
int blocks=0;

void loop() {
 sprite.fillSprite(TFT_WHITE);
 sprite.setFreeFont(&Orbitron_Light_24);
 sprite.drawString("Progress:",75,16);
 sprite.setFreeFont(&Orbitron_Light_32);

 

 sprite.drawString(String(progress)+"%",75,54);

 progress++;
 if(progress==101)
 progress=0;

  blocks =progress/5;
 sprite.drawRoundRect(x,y,w,h,3,TFT_BLACK);
 for(int i=0;i<blocks;i++)
 sprite.fillRect(i*5+(x+2)+(i*1),y+4,5,10,TFT_BLACK);
 sprite.drawRect(10,124,60,22,TFT_BLACK);
 sprite.drawRect(76,124,60,22,TFT_BLACK);
 sprite.drawString("OK",40,134,2);
 sprite.drawString("CANCEL",106,134,2);
 sprite.setTextFont(0);
 sprite.drawString("Installation almost done!!",80,160);
 sprite.pushSprite(0,0);
delay(60);
}

This is the error:
[... 331 lines cut ...]
c:\Users\david\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::drawPixel(int32_t, int32_t, uint32_t)':
c:\Users\david\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\david\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3677:7: note: in expansion of macro 'DC_C'
 3677 |       DC_C; tft_Write_8(TFT_CASET);
      |       ^~~~
c:\Users\david\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\david\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3684:7: note: in expansion of macro 'DC_C'
 3684 |       DC_C; tft_Write_8(TFT_PASET);
      |       ^~~~
c:\Users\david\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\david\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3690:3: note: in expansion of macro 'DC_C'
 3690 |   DC_C; tft_Write_8(TFT_RAMWR);
      |   ^~~~

exit status 1

Compilation error: exit status 1```

I appreciate any ideas.

lewisxhe, 2023-10-12:
Depending on the situation, there is a problem with your compilation environment. Please delete esp and reinstall it.

Davidhndrsn70, 2023-10-12:
If you are referring to removing and reinstalling esp32 by Espressif. I just tried that again with the same results.

[image]

lewisxhe, 2023-10-12:
[image]
Everything is normal here. It seems that the cause of the error is environmental problems. If you compile the examples that come with arduino esp32, such as WiFiScan, will it be normal?

Davidhndrsn70, 2023-10-12:
This is the only example code that I have used and it has worked perfectly up until the other day when this all started. When I try code with for example : 
#include <WiFi.h>
#include <WebServer.h>
#include <WebSocketsServer.h>
#include <ArduinoJson.h>

It works as soon as I add
#include <TFT_eSPI.h>

I get this loge error message..

assadaria, 2023-10-13:
Please figure out a solution since I've been experiencing the same error in my Arduino IDE since yesterday. I'm thinking it might be related to updating some libraries, such as esp32 from Expressif.

lewisxhe, 2023-10-13:
I updated to 3.0.0 alpha2 and did encounter compilation errors, but they were not the ones you encountered.

[image]

lewisxhe, 2023-10-13:
[image]

I also encountered the same error. It seems that esp 3.0 is a major update, which makes many examples unusable. Please keep using versions below 3.0 and wait for the upstream library version to adapt before updating.

closed 2024-03-22 by github-actions[bot]