Xinyuan-LilyGO/T-Display-S3 #204: Flash original firmware - black screen

asiorgad, 2023-11-28:
I played a bit with the board and flashed one example,
The screen became black after that, I've tried to restore to the original firmware , but had no luck, screen is still black,
Also I tried to look for firmware on circuitpython.org but the board isn't there . 
How can I restore the original firmware that came with the board?

teastainGit, 2023-11-28:
Looks like:
Solutions to common startup issues:
A) Unit seems "bricked" 
   	LCD does not display, unit remains dark with no white backlight:
	In declarations add 
		#define PIN_POWER_ON 15  // LCD and battery Power Enable
		#define PIN_LCD_BL 38    // BackLight enable pin (see Dimming.txt)
	Then in setup add
		pinMode(PIN_POWER_ON, OUTPUT); 	
		pinMode(PIN_LCD_BL, OUTPUT); 	
		digitalWrite(PIN_POWER_ON, HIGH);)	
		digitalWrite(PIN_LCD_BL, HIGH);)	

These two outputs must be 'ON'.
Solutions to common startup issues:
A) Unit seems "bricked" 
   	LCD does not display, unit remains dark with no white backlight:
	In declarations add 
		#define PIN_POWER_ON 15  // LCD and battery Power Enable
		#define PIN_LCD_BL 38    // BackLight enable pin (see Dimming.txt)
	Then in setup add
		pinMode(PIN_POWER_ON, OUTPUT); 	
		pinMode(PIN_LCD_BL, OUTPUT); 	
		digitalWrite(PIN_POWER_ON, HIGH);)	
		digitalWrite(PIN_LCD_BL, HIGH);)	
I cover this at my LilyGO T-Display S3 GitHub repository here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 

The firmware is permanently stored in ROM and cannot be destroyed!
The demonstration sketch that it came with is here:
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/factory
But it can be tricky to setup and compile. 
My sketch 'HotHead' is very simple and easily compiled.
-Terry

asiorgad, 2023-11-28:
Seriously,bricked?? I just used it for the first time. And i Already tried the steps you gave me.Im asking for a full refund for this board!If I’m not getting a refund:I’m in some WhatsApp's and telegram esp32/ardunio  group’s with thousands people, I’ll convince them not to buy anything from lillygo .And of course this is my last purchaseBy the way, I heard the same issue from multiple people , if you just selling crap “your good name “ will soon be gone.‫ב-28 בנוב׳

‬

https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/factory

lewisxhe, 2023-11-28:
Here are the factory firmware and writing methods, which can be used to quickly determine if there is a hardware problem
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/firmware/README.MD

Allmoz, 2023-12-12:
i had a similar issue. They might look very similar, but i actually got a AMOLED series... so this was the wrong repo, the things in https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series worked for me :)

closed 2024-03-21 by github-actions[bot]