Xinyuan-LilyGO/T-Display-S3 #205: Power T-Display-S3 with a external 3.3V source

danielcharrua, 2023-11-30:
Hello,

I need to power the board with a regulated external 3.3V source:

- Is ok to connect it trough `3V PIN` and `GND`?
- Sometimes a USB (PC) connection must be made to upload firmware, what happens when the two power sources are connected at the same time? If this is bad, using a Schottky diode on the external power source can solve this issue? I need to be sure that both devices are safe (board, external power source and PC)

[I have read that when using external power some pins must be changed](https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples) @teastainGit

Thanks. All help is much appreciated.

teastainGit, 2023-11-30:
Hi! teastainGIT here.
Due to the sophisticated power section of the LilyGO T-Display S3 the power incoming has to be 5V USB (4.2V may work) and 3V to 6V coming into the battery charge connector (below and facing backward of the USB-C).
USB 5V is available on one external pin as an output only to supply some sensors with 5V, but cannot be used to backfeed 5V to the board.

No pins must be changed when using external power. Two very important GPIOs must be set HIGH in order for the display and serial monitor to work. 
From my repository :
```
 #define PIN_POWER_ON 15  // LCD and battery Power Enable
  #define PIN_LCD_BL 38    // BackLight enable pin (see Dimming.txt)

just before the setup and then in setup add:
  pinMode(PIN_POWER_ON, OUTPUT);  //triggers the LCD backlight
  pinMode(PIN_LCD_BL, OUTPUT);    // BackLight enable pin
  
  digitalWrite(PIN_POWER_ON, HIGH);
  digitalWrite(PIN_LCD_BL, HIGH);
```
With the charging connector you can plug and unplug USB-C at random!
Hope this helps!
-Terry

danielcharrua, 2023-11-30:
Hey @teastainGit thank you for the response.

Correct me if I understand wrong: I can feed from 3V to 6V from an external power source into the battery charge connector?

And also I can plug the USB-C and load firmware? No need for diodes or any extra protection? No backfeeds or damaging USB-C port on the PC or the external power source?

Thank you again.

teastainGit, 2023-11-30:
You are correct.
The reason for the complicated design is to allow 5V from the USB to be fed to an IC battery charge controller chip and charge up the battery. I use a 5V white iPhone charger to power it, or 3 AA cells through the 2pin charge connector.
You can plug it into a computer USB and program it while charging off the USB!
Do not exceed 6V on either connector. 
Trying to connect 5V through the 5V pin on the board has resulted in reports of damage and/or faulty running. It is intended to power older 5V sensors from the Arduino days. It DOES backfeed with potential damage and upset some users.
The two outputs that I mention must be set HIGH with every sketch, even some cool app for ESP32 downloaded from somewhere on The Internet!
I am just an experienced user, not with LilyGO,  but I had tremendous problems with this board...so I made a GitHub repository.
-cheers

lewisxhe, 2023-12-01:
You can see from the schematic diagram that 5V = USB 5V. If the voltage is normal, then it has the same effect as connecting to USB.

lewisxhe, 2023-12-01:
They are the same line without any isolation. It is best not to connect them at the same time.

danielcharrua, 2023-12-02:
Hi @lewisxhe, great information. 

Is there a way to have both power supplies at the same time? I need to use the board with a power source always connected from 3V or 5V and sometimes connected to USB to update firmware. Do you have some information on isolating and having no issues? Any data is welcomed for me to investigate.

Thank you.

danielcharrua, 2023-12-03:
@lewisxhe Maybe I did not explain myself as good as I wanted... I'm installing the Display S3 into a coffee machine, and need to power the Display S3 with power gathered from the coffee machine main board (I think I can get 5V or 3.3V somewhere).

The Display S3 will also need to be connected to the USBC sometimes for some updates, so in that time, the Display S3 will be sourced with two power lines, the one coming from the coffee machine board and the USBC. I wanted to create some sort of protection and be safe with all devices.

lewisxhe, 2023-12-05:
Do you need to provide 3.3V or 5V from the coffee machine to provide power to DisplayS3?
You can directly connect 5V from the coffee machine to the 5V Pin on DISPLAY S3, and then cut off the VBUS 5V on the USBC line, so that you can update via USB and use 5V power supply at the same time

danielcharrua, 2023-12-05:
When you say "cut the VBUS 5V on the USBC line" do you mean to cut the +5V cable on the USB cable or to cut the trace on the circuit?

lewisxhe, 2023-12-05:
I think cutting the +5V cable is the simplest and safer way.

danielcharrua, 2023-12-05:
I agree but this means that the coffee machine must be on to upload firmware. This is a valid solution but not the one I'm looking for.

I would love that both cases can coexist and be protected from each other, using standard USB cables.

Imagine someone that builds this coffee machine and needs to remember to cut the USB and use that cable each time for uploading firmware while the coffee machine is turned on. This is difficult to remember and probably would use a normal cable sometimes and blow something... That's not a good outcome for the project 🤣

I know that introducing a Schottky diode on the external power supply will prevent that when the USBC cable is connected, no current (from the PC) will flow to the external power supply, the case I need to solve is how to protect the USB port of the computer when the external power is ON. 

The ideal would be that when the two power supplies are connected, external 5V from the VIN pin and USBC cable, the USBC takes priority and all devices are fully protected from reverse current.

lewisxhe, 2023-12-05:
Of course, this is possible. You can connect a Schottky diode in series to the 5V pin of the USB cable and a Schottky diode in series to the 5V pin. This is relatively safe.

danielcharrua, 2023-12-08:
Hello @lewisxhe,

So just to summarize, I have placed a Schottky diode before the VIN pin, so now when the USBC is connected no current will flow to the external power supply. The first part is done.

For the second part (protecting the PC from external current) I have some difficulties yet. 
I need to confirm that the T-Display-S3 has no Schottky diode to protect the PC. Is this protection is non existent on the actual design I need to place a Schottky diode between the PC and the USBC port, is that correct?

Is only me that is trying to achieve this? Connecting external power supply and USBC to the T-Display-S3 at the same time? Maybe would be nice to have this protection placed in the board design for other people trying to connect the T-Display-S3 from external power and upload firmware without having any risk of current flowing onto the PC or the external power supply.

I'm working in a project that transforms coffee machines into "smart" coffee machines using the T-Display-S3 and need to solve this. This project will be used by people without knowledge and I need the project to be safe, to protect both ends, the power supply part and mostly important the PC of people uploading firmware.

Thank you for your comments.

lewisxhe, 2023-12-09:
Maybe you can add a USB power isolator on the USB side

danielcharrua, 2023-12-10:
Hey @lewisxhe I've found a Schottky between VBUS and VIN already on the T-Display-S3 board. Can you confirm that the board is already protected with this diode and no current will flow from the VIN (when connected) to the USB port? Maybe no isolator is needed...

[image]

lewisxhe, 2023-12-11:
Oh! Yes, I didn’t see it either. Then it can be used without an isolator.

danielcharrua, 2023-12-11:
Great, so by placing a extra Schottky diode between the external power source and the VIN pin the T-Display-S3 could be powered and both power supplies (external & USB) will be protected from reverse current or damage.

closed 2023-12-11 by danielcharrua