Xinyuan-LilyGO/T-Display-S3 #207: Can't verify the sketch.

matmo1, 2023-12-08:
Hi,
I recently purchased a T-Display S3 and I use Arduino IDE 2.2.2.
I've decided to compile a sample project for the T-Display S3 using Arduino IDE from the example folder.
When i clicked verify it stopped and gave me this error:

\Documents\Arduino\T-Display-S3-main\example\PCBClock\bootloader.bin was unexpected at this time.

exit status 1

Compilation error: exit status 1

Also installed the most recent library.

lewisxhe, 2023-12-08:
Uninstall the esp board and then reinstall it

matmo1, 2023-12-09:
I reinstalled the board, reinstalled Arduino ide, reinstalled the library and I'm still getting the same error.

lewisxhe, 2023-12-09:
This is an environmental issue, not an example issue. I recorded a usage [video](https://www.youtube.com/watch?v=PgtxisFvMcc) before.

teastainGit, 2023-12-09:
Maybe this is a bug in the Nightly Build Arduino IDE 2.2.2?
Can you try uploading a simpler sketch like the tft example?
Is your boards version 'Arduino ESP32 2.0.13' and the upload type 'ESP32S3 Dev Module'?

matmo1, 2023-12-09:
It might be. I did downgrade the library and no luck. I tried the tft example and I got the error again. I'll install the 2.2.1 version and try again.

matmo1, 2023-12-09:
It worked! Thanks everyone!

closed 2023-12-09 by matmo1