Xinyuan-LilyGO/T-Display-S3 #208: Blank screen with TFT_eSPI

d3mac123, 2023-12-12:
I have 2 T-Display S3 boards and all of them were working fine with the TFT_eSPI library. However, yesterday, for some reason, both stopped to display (the code is working behind, as I can see via the Terminal messages). I have tried everything:
- Installed the non-touch firmware (all boards were able to display the firmware code) but, after trying my code, nothing displays anymore;
- then I looked at this thread https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/151 - it did not work as well;
- then I deleted the TFT_eSPI folders (in both /Library/Arduino15/libraries and Documents/Arduino/libraries - still no luck;

Any ideas on how to get my boards displaying again?

teastainGit, 2023-12-13:
Couple things to check:
1) In my repository (https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples ) there is a copy of the 'User_Setup_Select.h' file. It must copied to the  TFT_eSPI library folder. It is reset to default every time you allow the Arduino IDE to do an update. If there is no image, copy and replace this file.

2) Did you load my HotHead example? Did it work as expected?
3) Are you on battery or USB power?
4) Is the backlight on but no image?
These are the most common problems, let me know how you make out!
-Terry 
teastain@me.com

d3mac123, 2023-12-13:
@teastainGit Thanks for the help. Your HotHead example works fine. However, mine does not. I always run on USB power, btw. Backlight is on, terminal messages working as expected. Only the display is fully black :(

teastainGit, 2023-12-13:
2.5.4 and all further are LilyGO compatible (!) after a 6 month long struggle! But you ALWAYS have to edit the 'User_Setup_Select.h'  file as shown below:
```
comment out: //#include <User_Setup.h>           // Default setup is root library folder
un-comment: #include <User_Setups/Setup206_LilyGo_T_Display_S3.h>     // For the LilyGo T-Display S3 based ESP32S3 with ST7789 170 x 320 TFT
```

Your sketch must contain these components:
"If you are NOT running one of my sketches, you should add:"
```
  #define PIN_POWER_ON 15  // LCD and battery Power Enable
  #define PIN_LCD_BL 38    // BackLight enable pin (see Dimming.txt)

  just before the setup and then in setup add:
  pinMode(PIN_POWER_ON, OUTPUT);  //triggers the LCD backlight
  pinMode(PIN_LCD_BL, OUTPUT);    // BackLight enable pin
  
  digitalWrite(PIN_POWER_ON, HIGH);
  digitalWrite(PIN_LCD_BL, HIGH); 
```

d3mac123, 2023-12-13:
Thank you so much for your help here. Although my previous code is still not working (there is something I will have to figure out), I made a few simple examples and I can see the board is working fine. I really appreciate the help!

closed 2024-03-21 by github-actions[bot]

hamabao413, 2024-07-10:
> Couple things to check:
> 
> 1. In my repository (https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples ) there is a copy of the 'User_Setup_Select.h' file. It must copied to the  TFT_eSPI library folder. It is reset to default every time you allow the Arduino IDE to do an update. If there is no image, copy and replace this file.
> 2. Did you load my HotHead example? Did it work as expected?
> 3. Are you on battery or USB power?
> 4. Is the backlight on but no image?
>    These are the most common problems, let me know how you make out!
>    -Terry
>    [teastain@me.com](mailto:teastain@me.com)

Hi I had try your HotHead example but I got backlight on but no image and if i use others code I got no backlight and image
Please help thank you.