Xinyuan-LilyGO/T-Display-S3 #211: Unable to blink the onboard LED (

vbguyny, 2024-01-01:
I am using the T-Display-S3 (TOUCH): https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/image/T-DISPLAY-S3-TOUCH.png
I have updated to the flash file that was uploaded on Nov 19, 2023.

I am trying to get the following example working with the board with the onboard LED (there are two, one green and the other red): 
https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink

I don't see any where which GPIO pin these LEDs are on. Could someone please indicate how I can blink the onboard LED using VSCode and Expressif?

teastainGit, 2024-01-01:
They are un-blinkable!
Red is Battery charge voltage available.
Green is 3.3V OK.
-Terry

vbguyny, 2024-01-02:
@teastainGit Ah.... got it, thanks for the explanation.
I guess it is time to solder the header pins and connect it to a breadboard 😉

closed 2024-01-02 by vbguyny