Xinyuan-LilyGO/T-Display-S3 #212: Firware_no_Touch is the same as Touch ?

Maddin4000, 2024-01-02:
hello, i have different Firmware flashed but i can not change wifi, display show wifi networks with numbers before but i dont can switch without touch

teastainGit, 2024-01-02:
Exactly what firmware are you using?
Does the message show "EspTouch something, something"
Arduino IDE ?

Maddin4000, 2024-01-02:
T-Display no Touch bin
Flash Tool for windows
show message esp touch, but don't find  the esp
it will connect to xiang....

teastainGit, 2024-01-02:
That sounds like the Factory program. It tries to use "ESP Touch" which is a cell phone app to use your phone WiFi login to communicate with LilyGO and has NOTHING to do with LCD touch function.
If you use Arduino IDE you can edit the program with your home WiFi credentials and login. 
All this app does is demo a clock!!!
What do you hope to do with your LilyGO T-Display S3 once you get it sorted?

Maddin4000, 2024-01-03:
I like to experiment with it, not one of the Examples functions with my TDisplay S3, only I can run Nerdminer V2 on it. 

I run on Mac with Arduino IDE and Platformio, I can many Examples install but Screen remains empty. 

I am new in all of that, and I think I make one mistake so that’s runs nothing for me. 

I hope you can help me with a sample code for Arduino so that I can build on it?

Maddin4000, 2024-01-03:
LILYGO T-Display-S3 ESP32-S3 1.9 Zoll ST7789 LCD Display

teastainGit, 2024-01-03:
Great, I run Arduino IDE 2.2.1 on a Mac!  I had trouble at first so I wrote down all my issues into a GitHub repository:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
My HotHead examples are for de-bugging and the code is tutorial.
Burning the Factory firmware is a good way to check the hardware!
The library files  from the official LilyGO T-Display S3 repository called "Lib" MUST be copied into your libraries and overwrite and replace the existing Arduino IDE libraries. You cannot do a Arduino IDE library update or all is lost!
After you get my demo working you can let the TFT_eSPI Arduino IDE library update, but the only different file is called "User_Setup_Select.h" this is the only really LilyGO T-Display S3 specific, so I leave a copy on my repository that you can easily copy it back to the  TFT_eSPI library and  overwrite the "standard" file after an update.
Confusing? YES!
-Terry

Maddin4000, 2024-01-03:
Great now its working, i forgott to replace the User_Setup_Select.h

thx thx thx

closed 2024-03-21 by github-actions[bot]