Xinyuan-LilyGO/T-Display-S3 #213: With Sensirion SCD41, only got "Received NACK on transmit of address"

domotiquefacile, 2024-01-03:
Hello,

I am trying to use a T-Display-S3 to monitor a Sensirion SCD41 (CO2 sensor).

Instead of using JST, I connected/soldered SDA to GPIO43 and SCL to GPIO44.
According to this issue, I thought that it will work:
https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/85

I configured my arduino software using this setup:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples/blob/main/T-DisplayS3_Setup.txt
And I confirmed the setup by running the included "Hot Head" example, and it worked.

I then tried to test the SCD41 with this:
https://github.com/Sensirion/arduino-i2c-scd4x

Unfortunately I only got the error messages:
```
Error trying to execute getSerialNumber(): Received NACK on transmit of address
Error trying to execute startPeriodicMeasurement(): Received NACK on transmit of address
Waiting for first measurement... (5 sec)
Error trying to execute getDataReadyFlag(): Received NACK on transmit of address
```

Trying another example "Example8_SCD4x_BLE_Gadget_with_RHT.ino" from here:
https://github.com/Sensirion/arduino-ble-gadget

I got a little further:
```
Sensirion GadgetBle Lib initialized with deviceId = 26:19
Error trying to execute startPeriodicMeasurement(): Received NACK on transmit of address
Waiting for first measurement... (5 sec)
CO2(ppm)	Temperature(degC)	RelativeHumidity(percent)
Error trying to execute readMeasurement(): Received NACK on transmit of address
```

What could be wrong in my setup ?

Thanks a lot in advance

teastainGit, 2024-01-03:
OK, two things here
1) The app was written for the old Arduino boards and the Arduino IDE  can detect what EXACT board you are using! It picks the "correct" pins and quietly starts Wire in the background.

2) In the wild ESP32 world you need to add the SDA, SCL into the Wire.begin(43,44) if you are using the JST or whatever pins you chose to use.
-Terry

domotiquefacile, 2024-01-03:
Thanks a lot @teastainGit, I changed to Wire.begin(43,44) and now I can see some values !

closed 2024-01-03 by domotiquefacile