Xinyuan-LilyGO/T-Display-S3 #214: Display black and unable to display anything

cahudson, 2024-01-08:
Have tried this on both a t-display s3 and a t-display s3 pro. I am unable to have anything displayed on the display and the display remains black. It is the following configuration:

1) Arduino IDE 2.2.1
2) libraries from the lib folder of the t-display-s3 github are in place
3) in the TFT-eSPI User_Setup_Select.h the 206 library line is uncommented
4) the TFT_eSPI version matches what is in the github of 2.5.0.

I have done the following and verified the following:
1) uninstalled and reinstalled the Arduino IDE
2) Removed all the libraries and reinstalled from the github and verified the TFT_eSPI
3) Have tried multiple of the TFT_eSPI examples including the TFT_Print_Test and TFT_Clock
4) Have verified that the ino files when uploaded are running on the t-display by doing some quick Serial printlns back to the arduino ide

Note this worked originally but appears to have stopped working after several months of coding in arduino ide on multiple projects and then coming back to the t-display s3. The arduino ide was upgraded along with many libraries installed. But all those were removed and the ide was uninstalled and reinstalled.

Anyone have any thoughts as to why the display is not showing or some things to try?  thanks

teastainGit, 2024-01-08:
This is the number 1 problem with this board. It offers amazing performance, colour display, etc, but it is tricky to keep running due to Arduino IDE library updates.
It looks like you are doing all the right things!
Having said that, perhaps check out my LilyGO T-Display S3 GitHub repository here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
 I keep a copy of the corrected 'User_Setup_Select.h' file on hand (in my repository) to be more convenient.
Every software update overwrites that file!!!
My examples are bare bones TFT_eSPI with serial output.
Let me know how you get on.
-Terry

Superberti, 2024-01-08:
Same setup - same problem here :-(

Is there any reason why the patch for the display is not included Bodmer's in TFT_eSPI?

teastainGit, 2024-01-08:
'User_Setup_Select.h'  file is included in bodmer's repository.
All useable configurations can be selected here.
line 132:
 #include <User_Setups/Setup206_LilyGo_T_Display_S3.h>     // For the LilyGo T-Display S3 based ESP32S3 with ST7789 170 x 320 TFT

Superberti, 2024-01-09:
OK, I just forgot the display enable pin. It is printed on top of the display table and not on the ride side as the other signals ;-(
But now the display is up an running.
But you still need to use Lilygo's TFT_eSPI repo, not Bodmers.

Superberti, 2024-01-09:
OK, I was wrong, I tested again and I can confirm that the current Bodmer's TFT_eSPI is compatible with the display.
It's a little bit tricky as Arduino has no "clean all" function and the (already) compiled libs are cached. So before testing a different lib (by copying the code with the explorer) it is necessary to switch to a different board in order to clean up the caches.

So in the end it is not too hard to use the display:

1. Uncomment the right user_setup line (as stated in your previous comment)
2. Remember to switch on the display power and backlight before doing anything else.

Thanks and bye,

lv333ming, 2024-01-23:
> 已在 t-display s3 和 t-display s3 pro 上尝试过此操作。我无法在显示屏上显示任何内容，并且显示屏保持黑色。其配置如下：
> 
> 1. Arduino 集成开发环境 2.2.1
> 2. t-display-s3 github 的 lib 文件夹中的库已就位
> 3. 在 TFT-eSPI User_Setup_Select.h 中，206 库行未注释
> 4. TFT_eSPI 版本与 2.5.0 的 github 中的版本匹配。
> 
> 我已完成以下操作并验证了以下内容：
> 
> 1. 卸载并重新安装 Arduino IDE
> 2. 删除所有库并从 github 重新安装并验证 TFT_eSPI
> 3. 已尝试多个 TFT_eSPI 示例，包括 TFT_Print_Test 和 TFT_Clock
> 4. 通过对 arduino ide 执行一些快速串行 printlns 操作，已验证上传的 ino 文件是否在 T 显示器上运行
> 
> 请注意，这最初是有效的，但在 arduino ide 中的多个项目上编码几个月后似乎已停止工作，然后返回到 t-display s3。arduino ide 已升级，并安装了许多库。但所有这些都被删除了，ide被卸载并重新安装。
> 
> 有人对为什么显示屏不显示或需要尝试一些事情有任何想法吗？谢谢

I have the same problem and don't know how to solve it.

lv333ming, 2024-01-24:
I also encountered the problem of not being able to display 

#include <TFT_eSPI.h> 

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sprite = TFT_eSprite(&tft);

void setup() {
  // put your setup code here, to run once:
  tft.init();
  tft.fillScreen(TFT_RED);
}

void loop() {
  // put your main code here, to run repeatedly:
}

Write test code, black screen.   Hope it can solve the problems encountered。

Superberti, 2024-01-24:
@lv33ming:

What about 1) and 2) in my last post?

lv333ming, 2024-01-26:
> @lv33ming:
> 
> What about 1) and 2) in my last post?

Didn't solve the problem, could you elaborate on how to solve such a problem.

teastainGit, 2024-01-26:
So, this board is very picky about settings and libraries!
That is why I made a user support GitHub here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
Almost every time I start a LilyGO T-Display S3 project I have to down load the LilyGO repository and replace, overwrite, the Arduino IDE TFT_eSPI and lvgl library. 
I suggest that you read my repository and try uploading the HotHead example. It is simple but thorough sketch for  TFT_eSPI and it has a changing serial output for function check. 
A sketch with no functions in the main loop may jam up the CPU and prevent uploading. Always have at least 100ms delay in the main loop!
let me know how you make out. 
-Terry

lv333ming, 2024-01-28:
After replacing the file, an error occurred during compilation

In file included from C:\Users\Administrator\AppData\Local\Temp\arduino\sketches\93FB1A199AD77DC0883FEFC720D205D9\sketch\HotHead.ino.cpp:1:
C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32/Arduino.h:23:10: fatal error: stdbool.h: No such file or directory
   23 | #include <stdbool.h>
      |          ^~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

teastainGit, 2024-01-28:
Are you using Arduino IDE 2.2.1?
Are you using ESP32 board def 2.0.13?
LilyGO does nor recommend ESP32 ver 3 yet.
If so, can you try HotHead with board definition 2.0.13?

[image]

:also personally curious why your  directory slashes are \ back slash? Mine are / forward slash (M1 Mac)
I see 'esp32\3.0.0-alpha3\cores\esp32/Arduino.h' in your compile trace? forward and back slash!

lv333ming, 2024-01-28:
Success! I went directly to modify the file, and the compilation came through the screen.

lv333ming, 2024-01-28:
[image]

teastainGit, 2024-01-28:
Nice. 
Love success!
-Terry

OnlyFerris, 2024-02-19:
Received a T-Display-S3 in the mail today that I'd bought online

When I connected power, it displayed some stuff about trying to connect to a WIFI AP

I downloaded the ZIP file for this repo and extracted it in Arduino libs

Selected board "LilyGo T-Display-S3" in Arduino and tried to flash the "PokerS3" example.

Arduino IDE seemed to do this just fine

Screen is completely black now.

Not sure what I did wrong :S

OnlyFerris, 2024-02-19:
Tried the "PCBClock" example as well. Same result; black screen.

OnlyFerris, 2024-02-19:
My model is specifically "T-Display S3 AMOLED" and I just found that there is a separate repo https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED for that model so I am going to try those files now.

martinbertsch, 2024-03-16:
Would you be so kind to try to install this sketch and let me know if you get any display output? @teastainGit HotHead works fine. So I'm very clueless.

https://github.com/coppermilk/wiener_linien_esp32_monitor/tree/main

closed 2024-04-30 by github-actions[bot]