Xinyuan-LilyGO/T-Display-S3 #218: Battery and no display

vautieri, 2024-01-15:
With the 3.7 VDC batt plugged in, the red led goes bright, so I know it is connected.  However, when I disconnect rhe USB c, the display  turns off.   Do I really need to power this through the USB port for. The display?

teastainGit, 2024-01-15:
If you are new to the LilyGO T-Display S3 I suggest you look through my GitHub repository I made as a user.
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
Sounds like a case of:
"C) Can run on USB power but not battery connector:
	In declarations add
	#define PIN_POWER_ON 15		// just before the setup and then in setup add:
	pinMode(PIN_POWER_ON, OUTPUT); 	//"triggers the LCD backlight, and enables battery power
	digitalWrite(PIN_POWER_ON, HIGH);)	//on startup"
	
	(LilyGO variously refers to them as "LCD power on" OR "PIN_POWER_ON")
	Thanks to user RobLatour for finding this cure for the battery issue!"
-Terry

vautieri, 2024-01-15:
Worked like a charm and thank you for the quick response.

closed 2024-01-15 by vautieri

Seebit61, 2024-01-21:
Muchas gracias ha funcionado.Felicidades por su servicio. Ahora estoy investigando para que me diga la carga de la batería. Bartolomé Román

https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples