Xinyuan-LilyGO/T-Display-S3 #223: Low voltage on external lipo battery connected to the JST port

csthiang, 2024-02-04:
Board: t-display-s3 non-touch v1.2

I'm trying to power the board with a fully charged 3.7V lipo battery but it does not power on. I am aware of pulling pin 15 high for the display to work so I have the code bring up pin 15 and 38 for the screen. I've modified the `GetBatteryVoltage` demo code to add some sleep before powering the backlight on and it worked but I am seeing only 2.6-2.7v (Battery is at 4.18v as measured using multimeter). Measuring the 5v pin shows the same low voltage as well.

I have tried another of the same board and it is the same issue. There is no problem with using the usb-c to power the board. I have also updated the firmware but still seeing the same behaviour. 

What could be the problem? Faulty board (both???)?

teastainGit, 2024-02-05:
If you are new to LilyGO T-Display S3 you might try reading through my self-help repository :
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
The ESP32 has a terrible ADC and it is not stable or particularly linear.
You 'updated the firmware'
What firmware did you update? I'm not sure what that means, the bootloader is in ROM, any sketch you may upload is also technically 'firmware'!
-cheers, -Terry

closed 2024-03-21 by github-actions[bot]

Chihhao, 2024-11-03:
I encountered the same problem. Is there any solution?