Xinyuan-LilyGO/T-Display-S3 #225: Cannot find i2c devices via stemma/qwicc port

jaymedavis, 2024-02-13:
I have tried 3 different devices via the stemma port for i2c.  Downloaded the example code from this repo, which scans sda 18 and scl 17.

What am I doing wrong?  output is "No I2C devices found"

teastainGit, 2024-02-13:
The I2C JST Sh 1.0mm 4pin  port is SDA43, SCL44

[image]

I presume that you have a LilyGO T-Display S3 non-touch screen and NOT the LilyGO T-Display S3 AMOLED?
The I2C is the same but I like to clear that up.
I have my own User Support GitHub repository here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
This may be some help for you!
The  'HotHead_temp_sensor'  example is I2C.
-Terry

jaymedavis, 2024-02-13:
Awesome, thank you @teastainGit!  My I2C scanner is now finding my devices.  Starred/saved that repo as well!

closed 2024-02-13 by jaymedavis

Jelle7and9, 2024-12-15:
In addition, standard SDA and SCL pins are 18 and 17, not 43 and 44 as suggested in de image above. 
Quick google search brought me here and I was misled by this. Figured it out after a while :-)
You have to define SDA_PIN 43 and SCL_PIN 44 otherwise the connector is not I2C.