Xinyuan-LilyGO/T-Display-S3 #226: Compilation error: #error "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"

OnlyFerris, 2024-02-19:
Hi,

I'm using Arduino IDE version 2.3.1 on macOS

I extracted the ZIP file for this GitHub repo and put the directories from its `lib/` into `~/Documents/Arduino/libraries`.

When I started Arduino it asked me if I wanted to update libraries to latest versions and I selected yes.

I open the "PokerS3" example from this repo

When I hover over `#include <TFT_eSPI.h>` it shows the expected path

```text
/Users/user/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h
```

When I click the "Verify" button in Arduino IDE it tries to build but it fails saying:

```text
[...]
Generating function prototypes...
[...]
/Users/user/Documents/T-Display-S3/examples/PokerS3/PokerS3.ino:349:2: error: #error "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"
 #error  "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"
  ^~~~~
/Users/user/Documents/T-Display-S3/examples/PokerS3/PokerS3.ino:350:2: error: #error "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"
 #error  "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"
  ^~~~~
/Users/user/Documents/T-Display-S3/examples/PokerS3/PokerS3.ino:351:2: error: #error "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"
 #error  "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"
  ^~~~~
/Users/user/Documents/T-Display-S3/examples/PokerS3/PokerS3.ino:352:2: error: #error "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"
 #error  "Error! Please make sure <User_Setups/Setup206_LilyGo_T_Display_S3.h> is selected in <TFT_eSPI/User_Setup_Select.h>"
  ^~~~~
exit status 1
```

I then went and edited the file `/Users/user/Documents/Arduino/libraries/TFT_eSPI/User_Setup_Select.h` and uncommented this line:

```text
#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>     // For the LilyGo T-Display S3 based ESP32S3 with ST7789 170 x 320 TFT
```

but when I click the "Verify" button again I still get the same error

OnlyFerris, 2024-02-19:
I went back and replaced all of the extracted libraries again with the ones from the ZIP file, thinking that perhaps the updated versions were incompatible with the code from this repo

After doing that I am able to build the PokerS3 example.

Ideally this repo should be updated with latest versions of dependencies so that it does not happen

Let me know if you are open to Pull Request to do that

lewisxhe, 2024-03-05:
TFT_eSPI is constantly updated and cannot be synchronized. I have added how to update and troubleshooting instructions on the homepage. If you look carefully, you can find the solution.

closed 2024-04-19 by github-actions[bot]