Xinyuan-LilyGO/T-Display-S3 #230: T-Display-S3 Battery Charge  Circuit?

rpsavage, 2024-03-02:
I have several T-Display-S3 units and want to use them with small LiPo batteries that are rated at 350 mAH and 400 mAH. 

Before using them I wanted to check to make sure that the Li-Ion charging circuit is setup properly and would not over charge the batteries. Namely, the battery manufacturer specifies that the batteries should not exceed a 1C charge rate. For example, if the battery is rated at 350 mAH then you should not exceed 350 mA, and for the 400 mAH battery the charge rate should not exceed 400 mA. 

To check the T-Display-S3's charging circuit I looked at the schematic posted on GitHub and it shows that the charging circuit is controlled by U6 which is a TP4056 linear Li-Ion battery charger. The schematic also shows that a 2K ohm resister is used to program the charge rate. Using the TP4056 datasheet to calculate the charge rate it shows that if a 2K ohm resistor is used then the charge rate would be 580 mA which is too high for the batteries that I want to use.

Based on this I then looked into reprograming the charger for a lower rate by replacing the 2K ohm resistor with a 10K ohm resistor. Using a 10K ohm resistor would set the charge rate to 130 mA, which would be safe for the batteries. But when I try to find the TP4056 on the PCB it cannot be found? The datasheet shows that the IC should be a 8 pin SOP-8 package which cannot be located. Tracing out the PCB I find that the charging IC appears to be a device that is a 5-Lead TSOT-23 package with the marking of 65b311 printed on top. This does not agree with the posted schematic and I have the following questions and concerns?

1. Is it possible to get an updated schematic that shows the actual circuit for the T-Display-S3?

2. It would be good to provide instructions on how to replace the programing resistor (i.e. location of resistor and values for different charge rates). Providing the information would make the board more versatile and you would possibly sell more devices?

3. On the GitHub webpage it shows FAQ 10 "What battery to choose". The FAQ shows two examples, where 350 mAH and 400 mAH batteries are recommended. Based on the current schematic (TP4056 and 2K ohm resistor) the existing charge circuit would over charge the batteries. My research shows that using the existing circuit the battery capacity should not be lower than 580 mAH to be safely used. Please correct me if I am wrong or you may want to reevaluate the answer that is posted? 

4. To make sure that the T-Display-S3 is safely used, it would be a good idea to post the minimum size battery and charge rate in your specifications and include them on you main webpage.

Thanks

lewisxhe, 2024-03-05:
Hi , @rpsavage  

Please see the figure below for the location of the charging current setting resistor R13.
I asked and learned that TP4065 is used instead of TP4056

[image]

rpsavage, 2024-03-05:
The information that you provided is exactly what I need.

Thanks

nylar357, 2024-03-14:
So I have a question directly related to this one.  If I'm running my T-Display S3 off of and 18650 battery pack, when attached & USB-C cord applied with power am I charging the battery?  I see the RST light is lite up, but I want to be absolutely sure before I actually leave it for any extended period of time.  So to summarize :+1: 

- When USB-C connected and power + running the device off of an 18650 battery pack : Am I charging the battery or applying power from both sides?
- If charging to the battery is applied is this safe?
- Is there an easy way to get the battery level that I could possibly add to other sketches?

rpsavage, 2024-03-15:
Hi nylar357,
In my research and use of the T-Display S3 (somebody correct me if I am wrong) I have found the following,

**A. When USB-C connected and power + running the device off of an 18650 battery pack : Am I charging the battery or applying power from both sides?****

**Answer:** When the T-Display S3 is connected to both USB and a battery, USB will be providing power to everything and will also be charging the battery. The "RED" led will light up while the battery is being charged. And then when the battery is fully charged the "RED" led will go out. The battery only supplies power when the USB cable is unplugged.

**B. If charging to the battery is applied is this safe?****

**Answer:** In my case I consider charging to the battery to be "safe" when the charge current is less then or equal to the capacity of the LiPo battery. For example, if the I am using a battery rated at 400 maH then I would not charge the battery any higher then 400 ma. In your case to know if it is safe, you need to know the capacity of your battery and or ideally find out what the manufacture calls out. For example if the manufacture says keep the charge rate at 1C, then it means do not charge the battery at more then "1 x battery capacity".
 
Once you know the safe charge rate then you can decide if the battery charger circuit is safe to use. 
 
For the T-Display S3 the schematic shows that the battery is charged by U6 which is a TP4065. It also shows that a 2K ohm programing resistor is used (R13).  If you look up the data sheet for the TP4065 it provides the formulas used to calculate the programing resistor and also provides other information about charging. 
 
For me if I used the worst case  scenario (formula one in the ASIC Corp datasheet) then the battery charge current would be "1050/2K ohm = 0.520 amps" or 5200 maH.  So based on this I would only connect a battery for charging if the battery capacity is rated at least 0.52 amp hour. In your case you say that you are using some type of 18650 battery pack, so to know if it is safe you need to know if your battery's capacity is at least 5200 maH. 

**C. Is there an easy way to get the battery level that I could possibly add to other sketches?****

**Answer:** If you look at the schematic it shows that the battery "BAT" is connected to a voltage divider (R2 and R4) which is connected to "BAT_ADC" of the S3, which is really GPIO04. 

So using GPIO4 you can monitor the voltage of the battery. When the USB is connected you will measure about 5V/2 = 2.5 volts. Note, this what the battery will be charging at, it could be less because the charge voltage may vary as the battery is charged. 

When the USB cable is disconnected then you will measure what the battery voltage really is which would be "Battery Voltage/2".

When I measure the battery voltage I use the following,

**In the "void setup()" part of the sketch I would set the ADC resolution and attenuation using,**
   analogReadResolution(12);                       //Use 12 bits which would give range of 0 to 4095
   analogSetAttenuation(ADC_11db);           //ADC_11dB provide the most range, other setting ADC_0dB, ADC_2_5db, ADC_6db

**Then somewhere in the "void loop()" I would use,**
   int num = analogReadMilliVolts(4);          //Read the ADC voltage using the built in factory calibration, no conversion needed
   num = num * 2;                                        //Battery voltage is measured using a voltage divider  
   BatVolts = num;                                       //Actual battery voltage

closed 2024-04-30 by github-actions[bot]

fightforlife, 2025-01-30:
Is there a discharge protection cutoff?

rpsavage, 2025-01-31:
It been a long time since I looked at the schematic for the T-Display-S3, so I am not sure if the designer built in any discharge protection for the battery? 

What I do is use batteries with discharge protection built in. In this way if the batteries voltage gets too low then the built in circuit will prevent the lithium ion battery from going too low and  damaging it. Here's an example of a protected battery from SparkFun "**https://www.sparkfun.com/lithium-ion-battery-400mah.html"**. Product overview for the battery says that has "built-protection against voltage, over current, and minimum voltage." 

SparkFun has a good reference describing battery technologies and provides details related to lithium polymer/ion batteries batteries. "**https://learn.sparkfun.com/tutorials/battery-technologies**". In the section about "Charing and Discharging" they say the following.

> "LiPo batteries can also be harmed by being discharged too far. To protect against this, almost all LiPo batteries have a small safety circuit built into the top of the cell that will shut off the battery if the voltage drops below a certain threshold (usually 3V). "

What ever battery you use just make sure that the datasheet or manufacturer says that it has the built in protection. Note, you can also get 18650 cells with the protection built in.

lewisxhe, 2025-02-06:
No built-in battery protection, needs to be used with a lithium-ion battery with battery protection