Xinyuan-LilyGO/T-Display-S3 #233: T3S3 not able to install drivers on windows 11

franguyarmani, 2024-03-16:
Hi All,

I know this isnt exactly the right repo but there is not repo for the t3s3 and I think this board uses the same driver. I am unable to install the driver for the board on windows 11. When i run the installer for the driver I get the message that the driver was successfully pre-installed, but when i connect my device in flash mode, windows is still unable to use the driver for the device. Ive tried several different drivers, but none have worked. Am i correctly activating flash mode??

Thanks!

lewisxhe, 2024-03-16:
Win11 does not require any driver installation. Are you using T3S3? ? Or T-Display-S3? The method of entering download mode is the same for both of them, see README FAQ point 4

franguyarmani, 2024-03-16:
@lewisxhe I was just about to post the solution. If you use a usb-c to usb-c cable it doesn’t work. Needs to be usb-c to usb-a. 

Thanks for the quick response though!

closed 2024-03-16 by franguyarmani