Xinyuan-LilyGO/T-Display-S3 #236: Schematic for V1.2

Panometric, 2024-03-27:
The documents in [/schematic](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/schematic) both appear outdated. One is not versioned, the other is V1.1, but  V1.2 is what is being sold on the  [LilyGO Website](https://www.lilygo.cc/products/t-display-s3).  It's evident the charger IC has changed to a TP4065, a 5 pin part as noted in [Issue 230](https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/230), but other changes may have been made and they are not published. Please add an updated and versioned schematic to Github.

Panometric, 2024-04-28:
@Llgok , @lewisxhe can you get the latest?

lewisxhe, 2024-04-29:
There is no other difference except that the charging chip has changed.

closed 2024-06-13 by github-actions[bot]