Xinyuan-LilyGO/T-Display-S3 #237: Suddenly stop working

ortiztx, 2024-03-28:
I have bought this amazing original T-Display-S3 a couple of weeks ago.
I have been developing diffent projects using the Volos examples.
Today, after just two weeks, I tried to upload a code and serial was not responding.
I have tried to upload the firmware with the Flash download tool, but still is not able to connect with the board.
Alredy tried to press BOOT, RST, then release RST and BOOT, and no response.
I don't know if the reliability of the card is good or not, is strange to me.
Please, can you help on this?
Thnak you very much,
Tomas

teastainGit, 2024-03-28:
When I bought my first LilyGO T-Display S3 I made mistakes setting it up and did not understand the complexities so I made a GitHub repository as a resource here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
Hopefully you are using Arduino IDE 2.3.2?
Read through my repository and maybe see the answer yourself, which is more rewarding.
What was the very last sketch that you uploaded? Some people upload a 'empty' sketch to test, but this is too short and fast and the bootloader cannot stop it. Always include in 'Loop' a delay(100);
The button sequence is Boot, pause, RST clicked on and off, pause and then Boot released.
Alternately unplug the USB, hold in Boot and plug in and THEN release Boot.
Is your serial monitor showing any activity? My ' generic' tick tock serial sketch is good for that.
Let me know!

ortiztx, 2024-03-28:
Hey, Your magic and simple Tick Tock works fine, I have returned to control my T-Display S3.
Yes, I do use Arduino 2.3.2

Now I wanted to return to original Firmware, flashing by Flash Tool, and it cannot connect, I get this error:
Load_tool_3.9.5\bin
C:\Users\ortiztx\OneDrive - Abbott\Arduino\_Documentación\_Flash Download Tool Espressif\flash_download_tool_3.9.5\bin\firmware_no_touch.bin
[2024-03-28 19:59:12,072][ESP8266Loader_spi[1]][espDownloader.py][line:741][ERROR]:
no log file output ...

Then I tried the ESP Tool, connected to the port: [USB JTAG/serial debug unit (COM%): vinculado ] and I get this error:
oad_tool_3.9.5\bin
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting...Error: Failed to execute 'open' on 'SerialPort': Failed to open serial port.

The board has strange behaviours with the serial connection.

Do you have any other brigh idea to solve it?

Thank you again for your help,
Tomas

teastainGit, 2024-03-28:
'The original Firmware' is just a demo of little use. It does not contain the bootloader or any other needed firmware. The bootloader is hard printed in ROM memory and indestrucible!
So here is my order of solving comm issues with this board (with a known good sketch).
(I use the Arduino IDE connection window to determine if comm is OK.)

1) the boot, reset, boot sequence (watch the comm window in Arduino IDE )
2) unplug and hold Boot, plug in and release  (watch the comm window in Arduino IDE )
3) 'unplug and plug back in' is sometimes needed to start a new sketch
4) from the tools menu, click Erase All Flash before Downloading. 
(If that works, then disable that function for the next upload.)
5) Close and restart Arduino IDE 
6) restart your computer

Always monitor the upload  progress window and serial monitor window for signs of crashing, re-starting.

ortiztx, 2024-03-28:
Yeah, absolutely! As now it works fine, the factory image it was just to show my wife the clock it shows Lilygo in the advertising. I have also tried to ulpoad the factory.ino from the Git but again a lot of code issues, and unable to compile. It seems it's impossible to get the T-display-S3 with Clock again!!

Anyway, just one last question, I don't want to bother you more, I was looking into your repository and I have load several sketchs (I like the HotHead ;-)) and in the HotHead_DimmerNew.ino, I don't really understand how you are able to change the dim just pushing the button, because I see 14 different levels of dim, and in your loop you just puts LOW and HIGH the LCD pin. How is it possible that it changes without any loop or something like that?
Thank you again!!

void loop() {
  if (!digitalRead(topbutton)) {  //pressed is off
    digitalWrite(PIN_LCD_BL, LOW);
    delayMicroseconds(200);
    digitalWrite(PIN_LCD_BL, HIGH);
  }
  delay(200);
}

teastainGit, 2024-03-28:
So-o glad my repository helped. The HotHead sketch is also a TFT_eSPI programming tutorial, in that, I used simple easy to understand codes.
Good eye on the dimmer. It took a lot of research and false starts before realizing that the dimmer chip AW9364DNR is not PWM (ledC) like even Volos thinks (!), but very high tech string of closely timed pulses. 
Note the "delayMicroseconds(200);" which is a very short pulse, but it's IN THE SPEC SHEET!
The ledC PWM approach 'seems' to work but flakey and I was experiencing a unrequested dimming just after I reached full brightness, it drops down...
Good news!
Cheers, Terry

closed 2024-05-12 by github-actions[bot]

GaryInSanDiego, 2025-01-11:
A few things I noticed with the LILYGO T-DISPLAY-S3.
1) I was using ESP-NOW. I found it important to not do anything lengthy, especially to not send anything to the serial port while processing an ESP-NOW request. These requests are triggered by an interrupt and you want to keep this code short and sweet. I now set counters in global memory of the app, and check their values in a subroutine of the loop().
2) I switched from using an ACS712 sensor to an INA260 sensor (to measure current). The INA260 sensor was far more accurate because it had it's own VCC value to measure the current and also executes super fast compared to the ACS712. I found I needed to insert a delay(50) in my measurement loop to prevent the LILYGO display from rebooting every second. I ended up changing this to delay(200) (which results in 4 fps). Be careful how fast you're refreshing values on the display.