Xinyuan-LilyGO/T-Display-S3 #242: libs and platformio

jaymedavis, 2024-04-25:
is there a reason we have to copy around the "libs" folder to our projects?

in platformio, you can do like : 

lib_deps =
    bodmer/TFT_eSPI @ ^2.5.43

do you know the version #'s of the libs in this projects lib folder?  why can't we just define it like this instead of copying around those binaries to our own projects?

teastainGit, 2024-04-25:
Bodmer's TFT_eSPI  and LVGL need separate definition setups.
With TFT_eSPI  you can use the standard, latest TFT_eSPI  library freely from Arduino IDE or the GitHub repository...
you edit the 'User_Setup_Select.h' setup file to use settings for the unique 8bit parallel interface. I keep an edited copy in my Arduino IDE sketch folder and copy/overwrite the one in libraries/TFT_eSPI.
You can edit the existing User_Setup_Select.h but it is a bit tedious.
Comment-out this line:
//#include <User_Setup.h>           // Default setup is root library folder

and un-comment this one:
#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>     // For the LilyGo T-Display S3 based ESP32S3 with ST7789 170 x 320 TFT

The 'copy the libs' folder is more straightforward for new users.

jaymedavis, 2024-04-25:
Thanks so much for your comment.

I suppose I will keep a copy of `User_Setup_Select.h` for TFT_eSPI and overwrite it in my .pio/libdeps folder (possibly via script if I can figure it out).  

I assume you also had to do the same process for `lv_conf.h` for LVGL? 

Are all the other libraries out of the box?

teastainGit, 2024-04-25:
I mostly use TFT_eSPI  ! 
I use LVGL sometimes but take the lazy route and copy/overwrite.
There was a drama between LilyGO and Bodmer last year because they had to fork his library, but not re-name it. They worked with Bodmer to get the standard library to work with the 8bit parallel driver, settling on a device specific setup file.
The newer, much more expensive LilyGO T-Display S3 'AMOLED' version works straight out of the box, so, there's that!

08822407d, 2024-04-25:
> Thanks so much for your comment.
>
> I suppose I will keep a copy of `User_Setup_Select.h` for TFT_eSPI and overwrite it in my .pio/libdeps folder (possibly via script if I can figure it out).
>
> I [...]

TFT_eSPI's doc describe a way to assign configure for specific board if you work with platformio, by adding
`build_flags =
	-D USER_SETUP_LOADED=1
	-include $PROJECT_LIBDEPS_DIR/$PIOENV/TFT_eSPI/User_Setups/Setup206_LilyGo_T_Display_S3.h` to .ini

jaymedavis, 2024-04-25:
so this worked, thanks so much @08822407d.  The only thing im noticing on my project that changed, was the screen seems to flash 4 times off/on, before finally initializing.  leaves a pretty rough customer experience at the beginning.   

any idea what might be going on here?

jaymedavis, 2024-04-27:
disregard - the flashing was a problem with my code.  Thank you both so much again!

closed 2024-04-27 by jaymedavis