Xinyuan-LilyGO/T-Display-S3 #243: Using the display with LVGL & MicroPython

kdschlosser, 2024-04-25:
I have created an easy to use binding between LVGL and MicroPython. The "official" binding was too complicated to compile and get working properly and it didn't support I8080 or RGB busses. The one that I wrote does.

Here is the repository

https://github.com/kdschlosser/lvgl_micropython

The build is easy.  uninstall ESP-IDF if you have it installed into your system. The build system I wrote takes care of everything.
Clone the repo using

```
git clone https://github.com/kdschlosser/lvgl_micropython
```

you do not need to initialize any submodules or initialize the library at all. Just clone it and the build system will take care of the rest.

to compile..
```
cd lvgl_micropython

python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7789
```

That's all there is to it. It will take a while for the compiling to complete the first time around because of collecting all of the dependencies and setting up the build environment.

After the first time compiling you can use the following command to compile

```
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7789
```

once the build has finished you will be given directions at the end, there are 2 commands you need to run to flash the firmware. Run those commands in the order they appear. You might have to adjust the speed down to what is seen in the first command if it doesn't flash properly. This is due to the UART that is used on the board.

Once you have the firmware flashed unplug from the USB and plug it back in.

copy the code below and put it into a file called `main.py`

```

# compile command is
# python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7789

from micropython import const

_TP_RESET = const(21)
_SCL = const(17)
_SDA = const(18)

from micropython import const

_WIDTH = const(170)
_HEIGHT = const(320)

import lcd_bus

_DC_PIN = const(7)
_WR_PIN = const(8)
_DATA0_PIN = const(39)
_DATA1_PIN = const(40)
_DATA2_PIN = const(41)
_DATA3_PIN = const(42)
_DATA4_PIN = const(45)
_DATA5_PIN = const(46)
_DATA6_PIN = const(47)
_DATA7_PIN = const(48)
_CS_PIN = const(6)
_RD_PIN = const(9)

_FREQ = const(10000000)

display_bus = lcd_bus.I80Bus(
    dc=_DC_PIN,
    wr=_WR_PIN,
    rd=_RD_PIN,
    data0=_DATA0_PIN,
    data1=_DATA1_PIN,
    data2=_DATA2_PIN,
    data3=_DATA3_PIN,
    data4=_DATA4_PIN,
    data5=_DATA5_PIN,
    data6=_DATA6_PIN,
    data7=_DATA7_PIN,
    cs=_CS_PIN,
    freq=_FREQ,
    cs_active_high=False,
    pclk_active_low=False,
    pclk_idle_low=False,
    dc_data_high=True,
    dc_dummy_high=False,
    dc_cmd_high=False,
    dc_idle_high=False
)

fb1 = display_bus.allocate_framebuffer(int(_WIDTH * _HEIGHT * 2 / 10), lcd_bus.MEMORY_DMA | lcd_bus.MEMORY_INTERNAL)
fb2 = display_bus.allocate_framebuffer(int(_WIDTH * _HEIGHT * 2 / 10), lcd_bus.MEMORY_DMA | lcd_bus.MEMORY_INTERNAL)

import st7789

_RESET_PIN = const(5)
_POWER_PIN = const(15)
_BACKLIGHT_PIN = const(38)
_OFFSET_X = const(0)
_OFFSET_Y = const(0)

import lvgl as lv

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    reset_pin=_RESET_PIN,
    reset_state=st7789.STATE_HIGH,
    power_pin=_POWER_PIN,
    power_on_state=st7789.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7789.STATE_PWM,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(True)
display.init()
display.set_backlight(100)

scr = lv.obj(lv.screen_active())
btn = lv.button(scr)
btn.center()
label = lv.label(btn)
label.set_text('Hello World!')
label.center()

import task_handler

task_handler.TaskHandler()
```

upload the file to your board using ampy or mpremote. DO NOT use thonney at all. I do not know what it is about thonney but it seems to cause problems. Once you have the file uploaded unplug from the USB and plug it back in again. You should see something on the display.

This is going to give you the best performance that can be had when using MicroPython. It is using double buffering and SRAM DMA memory.

If you have any problems please open an issue on my repo.

lewisxhe, 2024-04-29:
Thanks, let more people see it.

kdschlosser, 2024-04-29:
This binding supports I8080, SPI and RGB on the ESP32 MCU's. RGB is only supported on the newer models. Supports using DMA memory on both SRAM and SPIRAM. You can use bounce buffers with the RGB to help reduce tearing, Double buffering is supported on all busses and all MCU's..

There is a display driver framework that makes it really easy to write a custom driver. The display drivers are written in Python which means they are runtime compiled so they are able to be swapped out without having to compile and flash new firmware. The same thing applies to touch interfaces and other types of input devices. keyboards, rotary encoders, mice, touch screens, etc...

This will work for any of the LilyGO ESP32's that have displays attached to them.

I just did a rework of the SPI bus if using a single frame buffer. It cuts the time to write the frame buffer to the display in half..

closed 2024-06-13 by github-actions[bot]

bobziuchkovski, 2024-11-18:
@kdschlosser How would your example above look for current main of https://github.com/kdschlosser/lvgl_micropython? It looks like the APIs have changed a bit. I can get the backlight lit on the T-Display-S3 but nothing rendering from lvgl to the display screen.

I'll keep tinkering but figured I'd ask in case you had a current example handy. I'm sorting through and trying to reconcile examples you have for the other chips vs what's on the T-Display-S3.