Xinyuan-LilyGO/T-Display-S3 #247: My board doesn't connect to WiFi

AlexC-0806, 2024-05-11:
I recently bought a T-Display S3 board and tried some default sketches. They work fine, but then I tried creating a sketch to connect the board to WiFi and get the time data from NTP. The board doesn’t connect to WiFi, even though I’m sure the WiFi credentials are correct. Here’s my code:

```
#include <Arduino.h>
#include <Arduino_GFX_Library.h>
#include "WiFi.h"
#include "time.h"

#define WIFI_SSID "MY_SSID"
#define WIFI_PASS "MY_PASSWORD"

#define LYLIGO_T_DISPLAY_S3
#define GFX_EXTRA_PRE_INIT()          \
  {                                   \
    pinMode(15 /* PWD */, OUTPUT);    \
    digitalWrite(15 /* PWD */, HIGH); \
  }
#define GFX_BL 38
Arduino_DataBus *bus = new Arduino_ESP32PAR8Q(
    7 /* DC */, 6 /* CS */, 8 /* WR */, 9 /* RD */,
    39 /* D0 */, 40 /* D1 */, 41 /* D2 */, 42 /* D3 */, 45 /* D4 */, 46 /* D5 */, 47 /* D6 */, 48 /* D7 */);
Arduino_GFX *gfx = new Arduino_ST7789(bus, 5 /* RST */, 0 /* rotation */, true /* IPS */, 170 /* width */, 320 /* height */, 35 /* col offset 1 */, 0 /* row offset 1 */, 35 /* col offset 2 */, 0 /* row offset 2 */);

void setup()
{
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
#ifdef GFX_EXTRA_PRE_INIT
  GFX_EXTRA_PRE_INIT();
#endif

#if defined(LCD_PWR_PIN)
  pinMode(LCD_PWR_PIN, OUTPUT);    // sets the pin as output
  digitalWrite(LCD_PWR_PIN, HIGH); // power on
#endif

#ifdef GFX_BL
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);
#endif

  // Init Display
  if (!gfx->begin())
  {
    Serial.println("gfx->begin() failed!");
  }
  gfx->fillScreen(RED);
  gfx->setTextSize(2);
  gfx->setTextColor(WHITE, PURPLE);
  gfx->setCursor(5, 5);
  gfx->println("Clock Test");
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  gfx->setCursor(5, 25);
  gfx->print("Connecting to WiFi...");
  while(WiFi.status() != WL_CONNECTED){
    delay(500);
    gfx->print(".");
    Serial.println("Connecting to WiFi...");
  }
  gfx->println("Connected to WiFi");
  configTime(3600, 0, "pool.ntp.org", "time.nist.gov");
  gfx->println(time(nullptr));
}

void loop(){
  gfx->fillScreen(WHITE);
  delay(500);
  gfx->fillScreen(BLACK);
  delay(500);
  gfx->fillScreen(RED);
  delay(500);
  gfx->fillScreen(GREEN);
  delay(500);
}
```

The loop() isn’t really doing anything because I wanted to test if the basic WiFi function would work. I’m using Arduino GFX because it’s the simplest library that works, since I wasn’t able to use tft_eSPI, but this isn’t the problem now. The problem is that the board doesn’t connect to WiFi, even though I tried the WifiScan sketch from the Arduino GFX library and that works, so the board is able to scan WiFi networks. The problem is that it isn’t able to connect. I bought this board with really high expectations, so it would be very disappointing if I weren’t able to use its WiFi capabilities.

teastainGit, 2024-05-12:
As a User I made this self-help GitHub repository here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
Just now I added a simple example of WiFi and NPT time.
Try it out !
-Terry

AlexC-0806, 2024-05-12:
> As a User I made this self-help GitHub repository here:
> https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
> Just now I added a simple example of WiFi and NPT time.
> Try it out !
> -Terry 

Thanks, but your code is made to be used in the Arduino IDE, but I'm using PlatformIO to program my board. I'll have to apply some changes to it to make it work in PlatformIO or simply I'll have to reinstall Arduino IDE and try it there. Maybe you could consider making a PlatformIO version of your repo. Thanks anyway!

teastainGit, 2024-05-12:
Perhaps you have a PlatformIO issue, not a _LilyGO T-Display S3 Issue_, as the code and product works as expected.

AlexC-0806, 2024-05-12:
> Perhaps you have a PlatformIO issue, not a _LilyGO T-Display S3 Issue_, as the code and product works as expected.

Well, I always used PlatformIO to program ESP32 and ESP8266 boards, and WiFi functionalities always worked without problems. Anyway, I think I'll try to adapt the code in your repo to PlatformIO and see if it works

AlexC-0806, 2024-05-12:
> Try it out ! -Terry

I tried to adapt that code to PlatformIO, importing the libraries directly from the official Lilygo repo, and adding lib_ignore in my platformio.ini file

[image]

I don't know why I get this error in compilation, both printLocalTime() and initWiFi() are declared in my code, since I literally made copy and paste from the code in Terry's repo.

[image]

lewisxhe, 2024-05-13:
Is your wifi functioning properly? Have you checked the wifi account and password? Are they correct?
You can ignore the above errors and compile directly to see if there are any errors.

AlexC-0806, 2024-05-13:
> Is your wifi functioning properly? Have you checked the wifi account and password? Are they correct? You can ignore the above errors and compile directly to see if there are any errors.

Yes, my wifi network is ok and my router doesn't block any devices, I double-checked the password and SSID and they are correct. I can't ignore the errors in compilations because the compilation stops when these errors appear. I even tried on different WiFi network and the board doesn't connect to them.

lewisxhe, 2024-05-13:
Using the example above, if you can't make the connection, I suspect it's a board problem. Contact your dealer and get a replacement board.

AlexC-0806, 2024-05-13:
> Using the example above, if you can't make the connection, I suspect it's a board problem. Contact your dealer and get a replacement board.

I already sent a mail to  <sales@lilygo.cc> asking for a refund or replacement, I hope they'll answer soon.

lewisxhe, 2024-05-13:
Okay, I've informed them

AlexC-0806, 2024-05-13:
Ok, thanks!

AlexC-0806, 2024-05-14:
Update: they answered asking me a video. I'll record it later, but for now I sent them the link of this issue so they can check it out. Still thanks to @lewisxhe and @teastainGit for the quick help! 😃

AlexC-0806, 2024-05-15:
Further update: I sent them the video 20 hours ago but still they didn't answer me. What could I do, apart from waiting?

lewisxhe, 2024-05-15:
I've told them and they will contact you soon

AlexC-0806, 2024-05-15:
> I've told them and they will contact you soon

Thank you so much!

AlexC-0806, 2024-05-16:
Further further update: Sorry for the "spam," and I don't mean to be argumentative or impatient, but it has been two days since I sent the email, and I have yet to receive any response. I understand that Lilygo has thousands more service requests to handle, but I think they have enough employees to be able to meet everyone's needs. Not that I am in a hurry, but this long wait does not bode well for me. I also know that maybe this is not the best place to file a complaint, would it be better to send another email?

closed 2024-06-03 by AlexC-0806

mo-kuh, 2024-07-07:
ESP32-S3 seems to have WIFI hardware problems: https://www.esp32.com/viewtopic.php?t=32109

Available WIFI networks can be listed (recieving wifi signals work) but no connection is possible (transmitting wifi signal might be the problem). It seems to be that the location of the antenna seems to be crucial for the ESP32-S3. To me it reads like the internal antenna is not operational. I wonder how this could be the case and the product still made it to costumers ...

I have 2 devices, non of them connect with your code @teastainGit (thanks for providing it though). 

I am still hopefull :) Anyone has any tips?

mo-kuh, 2024-07-07:
UPDATE: I could successfully to the following:

1) connect to an OPEN access point (no password)
2) Connect to the ESP32 in SoftAP mode

It seems like the issue is 
1) NOT related to transmitting (TX) signals 
2) related to the password / authentication mode

AlexC-0806, 2024-07-10:
> UPDATE2: Connection to 1 access point was successfull after setting "WiFi.setMinSecurity(WIFI_AUTH_WPA2_PSK);" "WIFI_AUTH_**WPA**_WPA2_PSK" was NOT working.

I tried this on my "broken" T-Display S3 board and it doesn't work. I even tried to set the rx power to the max, but it doesn't change anything. It seems that on my board that wasn't the issue, even because on the replacement board I got WiFi connection works fine without issues so I suppose that was an hardware problem, or another type of software issue.