Xinyuan-LilyGO/T-Display-S3 #248: Display turns off when calling tone()

wichjan, 2024-05-12:
I encounter the same issue previously described in the closed and unsolved issue #95:
When using the tone()-functionality (for example to run a buzzer) the display turns off.
For this to happen not even a real buzzer has to be connected to some i/o-port, just calling tone(_params_)
or oven just calling noTone(_some Pin-Nr_) immediately turns off the display (and backlight).

Does anyone have an idea how to solve this issue or where to look for the reason of this peculiar behavior?
Regards,
Jan

lewisxhe, 2024-05-13:
Tone calls channel 0 by default. Just change the following to channel 1.
```
   ledcSetup(1, 2000, 8); // Backlight
   ledcAttachPin(38, 1);
   ledcWrite(1, 200);
```

wichjan, 2024-05-13:
Thanks for the quick support!
I changed the pwm_channel to '1' in the backlight setup, but unfortunately it did not change the strange behavior …

lewisxhe, 2024-05-15:
There is a problem with your loop. It does not enter every 5 seconds at all, but continuously executes noTone, which causes the watchdog to trigger. The following is the repair procedure.
```

// https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/248

#include <TFT_eSPI.h>
#include "TFT_eSPI.h"

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite scr = TFT_eSprite(&tft);

void setup()
{
    Serial.begin(115200);
    pinMode(15, OUTPUT);
    digitalWrite(15, 1);

    ledcSetup(0, 2000, 8);  // Backlight
    ledcAttachPin(38, 0);
    ledcWrite(0, 200);

    tft.init();
    scr.createSprite(170, 320);
    scr.fillSprite(tft.color565(180, 180, 180));
    scr.pushSprite(0, 0);

} // end setup()

uint32_t t_interval ;
void loop()
{

    if (millis() > t_interval ) {
        Serial.println("noTone");
        t_interval = millis() + 5000;
        noTone(1); // kicks off display, same happens with any other pin
    }

} // end loop()

```

wichjan, 2024-05-15:
Yeah, I kept the Demo-Code a little too clean and forgot to reset the time-stamp `t_intervall` inside the `if`-block, so that after the first 5 seconds after booting, `tone()`is called in each `loop()`-pass. But why this should cause the board to reboot I really don’t understand. 
And this does not change with your code either.

**But:** if I call (like in a more realistic example) `tone(...)` before calling `noTone()` everthing works fine!
Also in the actual project-code, after changing the PWM-channel for the backlight to '1', as you suggested in your first answer! (I should have tried this earlier, but I focused on the example-code).
So, the case is solved :–) (although I did not understand what really happens).
Thanks again,
Jan

closed 2024-05-15 by wichjan