Xinyuan-LilyGO/T-Display-S3 #253: periodically resynchronise the internal Real time clock

paulharbison, 2024-05-27:
Arduino/esp32/v:2

Is there a way or function to periodically resynchronise the internal Real time clock.

My sketch I copied ( Rui Santos) only gets the time from Wi-Fi once and switches off the internet (as it’s not needed any more*)*.

Well it is needed! It gets out of sync by as much as 1 or 2 minutes per month.

Which is worse than cheap digital watch. The precise time from Network Time Protocol (NTP) is wasted here.

Maybe resynchronise the RTC once per hour would be sufficient.

Any method or relatively simple solution would be much appreciated.

teastainGit, 2024-05-28:
Hi, just a User passing by!
I set up a GitHub repository for problems setting up and some examples here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 

Check out my recent home-grown NTP time sketch for Arduino IDE .
The ESP32S3 has a terrible RTC, ADC and DAC, just marginally useful, but free!

I am considering adding the built-in RTC to keep time and only goto NTP once an hour, to keep internet traffic down, a small part I know…

(I am modifying it now as we speak so check it out and come back later for any updates.)

-Terry 
If you want to go offline of GitHub, here is my email:
***@***.***

Cheers !
Terry Stainton

***@***.***

paulharbison, 2024-05-28:
Thanks for your reply iwill look into it,

closed 2024-05-28 by paulharbison