Xinyuan-LilyGO/T-Display-S3 #256: Using the PCBDesignClock example compiles but there is not enough memory on device to run?

fromutome, 2024-06-01:
The example code PCBDesignClock compiles with Arduino IDE but after that the erro below say that program memory space is to large. I have not changed anything in the code.

Are there LilyGO T-display S3 devices with different size of memory?
Am using the LilyGO T-Display S3 in the board selection in the Ardiono IDE.
When using the board info function in arduino it says it's a ESP32-S3 Box?

Best Regards
//Johan

Sketch uses 3147449 bytes (100%) of program storage space. Maximum is 3145728 bytes.
Global variables use 47104 bytes (14%) of dynamic memory, leaving 280576 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

fromutome, 2024-06-04:
Thanks for the suggestion but that partition scheme is not availble on my installation/configuration.
How can I enable this option?

teastainGit, 2024-06-04:
They have screwed up with their new board def! 
Use ESP32S3 Dev Module board def and select 
Partition Scheme: "Huge APP (3MB No OTA/1MB SPIFFS)

[image]

fromutome, 2024-06-04:
Thanks! that solved the issue :)

closed 2024-06-04 by fromutome

akshayamytaneja, 2024-07-15:
[image]

Facing the same problem.
Using a T-Display S3 (no touch)

Upload procedure: Hold down BOOT button, press and release Reset, let go of the BOOT button (has worked for other examples)

teastainGit, 2024-07-15:
What 'ESP32 by Espressif' board def revision are you using?

whzbngry, 2024-07-15:
I think I haven't upgraded from 2.0.16 since it breaks my other projects.
[image]

teastainGit, 2024-07-15:
VOLOS! The guy is a genius, but he assumes we all are!
I had to watch his video frame by frame to learn the correct settings.
[image]
8MB and 8MB with SPIFFS
I use ESP32 ver 2.0.15, just my habit for now.
Cool app though!
-Terry

whzbngry, 2024-07-15:
Thanks a lot @teastainGit !
I got it working with the following settings with 2.0.16 esp32 by Espressif.

[image]