Xinyuan-LilyGO/T-Display-S3 #259: Max power allowed on the USB-C port

bartissimo, 2024-06-19:
Hi,

Can you please advise what the maximum power is you can put on the USB-C port (via USB cable) without damaging the T-Display-S3?

5V * ?A = how many watts max?

Thank you,

B.

lewisxhe, 2024-06-26:
There is no maximum requirement, only minimum requirements. A regular 5V 1A adapter is sufficient.

teastainGit, 2024-07-02:
Listen, You can use any high power supply as long as it is no more than 5VDC and supplies the board with no more than 1A, i.e. 5Watts.
Most ESP32 GPIO outputs are rated for no more than 20mA, and the total GPIO draw per chip is not more than 200mA.
The boards 5V and 3.3V pin traces may not be able to handle 1A without peeling off or fusing open.
Any large load, string of RGB, Motors, etc. should be powered from the source 'parallel' to the USB, with common grounds, not From the board.

closed 2024-08-16 by github-actions[bot]