Xinyuan-LilyGO/T-Display-S3 #266: T-DISPLAY-S3 PlatformIO and framework ESP-IDF monitor doesn't work

BullT, 2024-07-25:
Hi,

I'm using the LILYGO T-DISPLAY-S3 with PlatformIO and framework ESP-IDF.

When I upload and monitor, the monitor doesn't work.

Do you know why?

[image]

teastainGit, 2024-07-26:
Did you 'Enable' USB CDC on Boot?

BullT, 2024-07-26:
Hi,

I don't know how do it :)

teastainGit, 2024-07-26:
build_flags =
  -D ARDUINO_USB_MODE=1
  -D ARDUINO_USB_CDC_ON_BOOT=1

BullT, 2024-07-26:
Hi and thank you,

But it doesn't work.
I'm using framework ESP-IDF not Arduino.

[image]

BullT, 2024-07-26:
Hi,

I'm going to move it.

Thank you

closed 2024-07-26 by BullT

BullT, 2024-07-26:
Hi,

I found the concern.
It was on menuconfing, I had selected Default: UART0, after change to USB CDC it start to works:

[image]

teastainGit, 2024-07-27:
Very good news, thanks for getting back, I will add it to my experience.