Xinyuan-LilyGO/T-Display-S3 #270: Output voltage pins  5V, 3.3V, 3V

vldgeorgiev, 2024-07-31:
Hello

In the pinout of the T-Display-S3 there are two 3V pins and one 5V pin. It seems that the 3V pins are 3.3V, which is regulated. At least that's what I measured when powered through USBC. I haven't checked the voltage while powering from an almost discharged LiPo battery.

What I am interested is the purpose of the 5V pin. Somewhere in the docs or issues I saw that it is not intended to provide +5V power to the S3, only through USB or through the battery.

This comment https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/205#issuecomment-1834132748https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/205#issuecomment-1834132748 states the +5V is 

> "USB 5V is available on one external pin as an output only to supply some sensors with 5V, but cannot be used to backfeed 5V to the board."

However, it doesn't seem to provide a stable 5V output. I measured 4.2-4.5V on it. On the schematics (as far as I understand it) there is no up-converter that will provide 5V from the battery or USB voltage.
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/schematic/T_Display_S3.pdf

Can someone confirm what is the usage for the 5V pin?

Thanks.

lewisxhe, 2024-08-02:
5V is just a label. This is a label setting error by the hardware engineer. In fact, this is the end point of the path between the battery and USB. If the battery is not connected, you can directly input the DC 5V power supply here

If the battery is connected, this 5V label can directly output the battery voltage

Summary: This is a label error. The correct label should be VSYS, not 5V

vldgeorgiev, 2024-08-02:
Thank you. That explains it.

closed 2024-08-02 by vldgeorgiev