Xinyuan-LilyGO/T-Display-S3 #271: How to use GPIO43 as Input

BullT, 2024-08-01:
Hi,

I'd like use the GPIO43 in ESP32-S3 as a Input.

I'm doing this (I don't know what I have to put on <XX>):

`            gpio_iomux_in(GPIO_NUM_43, <XX>);
            gpio_config_t config{BIT(43),
                                 GPIO_MODE_INPUT,
                                 GPIO_PULLUP_ENABLE,
                                 GPIO_PULLDOWN_DISABLE,
                                 GPIO_INTR_POSEDGE};
            gpio_config(&config);
            gpio_install_isr_service(0);
            gpio_isr_handler_add(GPIO_NUM_43, button_handler, this);`

How Do I have to do to use the GPIO43 as Input?

Thank you

teastainGit, 2024-08-01:
GPIO43 is a GPIO and can be an input or an output.
(I just confirmed right now by a simple test)
This is the pinmap.
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/image/T-DISPLAY-S3.jpg

Your question would be better answered in a ESP-IDF forum, I think.

BullT, 2024-08-01:
Hi,

I found the solution:

        void init(void)
        {
            gpio_set_direction(GPIO_NUM_43, GPIO_MODE_INPUT);
            gpio_config_t config{(1ULL << 43),
                                 GPIO_MODE_INPUT,
                                 GPIO_PULLUP_DISABLE,
                                 GPIO_PULLDOWN_DISABLE,
                                 GPIO_INTR_POSEDGE};
            gpio_config(&config);
            gpio_install_isr_service(0);
            gpio_isr_handler_add(GPIO_NUM_43, button_handler, this);
        }