Xinyuan-LilyGO/T-Display-S3 #275: reboot at wifi

ericgrand, 2024-09-06:
My T display S3 has started to develop a bug. As soon as it connect to wifi, it reboots. I have reuploaded my sketch with no improvement.
I have used the flash tool and the firmware available on the github and same behaviour, I have the Lilygo logo, then scan wifi and it reboots.
Same behaviour on various USB cable and power source...
I suppose it is a harware issue?

Thanks for your support!

Eric

lewisxhe, 2024-09-09:
If it restarts when connected to wifi, please replace the data cable and power supply for testing. Test it under several power supplies. If the same result is obtained, it can be determined that there is a problem with the PCB power supply.

ericgrand, 2024-09-09:
Ok, so it is a defective PCB. It has been tested with different cables, power sources and even with external battery through the battery port.
Is there any warranty on this item? It is less than 18 months old.

lewisxhe, 2024-09-09:
I don't know if there is a warranty, you need to ask where you bought it

closed 2024-10-25 by github-actions[bot]