Xinyuan-LilyGO/T-Display-S3 #278: Factory Example and tone()

7c, 2024-09-20:
I have attached small speaker/buzzer to pin 13 (which seems to be not connected). In the given Factory example with lvgl. Whenever i call tone() function, the screen is getting black. next time its called screen comes back for a moment and then after calling the tone() it goes off. So once tone() called it stays black. Is this something todo with S3 or lvgl? I have also tried to generate the tone with https://github.com/weedmanu/ToneESP32/blob/main/src/ToneESP32.cpp which uses ledcWrite under the hood. Same problem. I hear the sound but the display is getting off. It seems not to be a current issue neither because if i try to tone the pin3 which has no load, i hear no sound but display acts same way. Please help me to find the problem.

7c, 2024-09-20:
i found that `ledcWriteTone(_channel, 0);` command inside `void tone_task(void*)` of Tone.cpp causes the screen is switched off. But regardless which pin i choose, have tried 3, 13,10 on S3.

lewisxhe, 2024-09-21:
Post your minimal test code

7c, 2024-09-21:
Ok [here](https://github.com/7c/T-Display-S3-Factory) is the latest FactoryCode stripped down to be without wifi, it sets an imaginary time, the buttons are bound to tone() and my_tone(), button1 will demonstrate the display goes off after the tone has been played, button2 will demonstrate it plays and display keeps working. 

The demo is at https://github.com/7c/T-Display-S3-Factory - it is tied to the s3 and has only minimal code and libraries. the my_tone is a fork of tone() with single line of code removal you can see at https://github.com/7c/T-Display-S3-Factory/blob/main/examples/factory/sound.cpp which basically comes from arduino-esp32 6.5.0. I have also tried with `platform = espressif32@6.8.0`. [Latest espressif32](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Tone.cpp) library seems to be slightly [different code](https://github.com/espressif/arduino-esp32/blob/7018cd114d00249674567846c9d67fbb3a1240a3/cores/esp32/Tone.cpp#L56) . The 6.5.0 library and 6.8.0 seems to be using `ledcWriteTone(_channel, 0);` and latest Tone.cpp seems to be using `ledcWriteTone(tone_msg.pin, 0);`

lewisxhe, 2024-09-23:
The method of ledc controlling display-S3 is wrong. I have modified it to the correct control method. Please see the commit information

7c, 2024-09-23:
i confirm it works, thanks for following up, i love your products even more now.

closed 2024-09-23 by 7c