Xinyuan-LilyGO/T-Display-S3 #279: PlatformIO installation instructions not working.

kamikaze2112, 2024-10-01:
I've been at this for hours now trying to sort out how to install this repo in PlatformIO and can't for the life of me get it working.

I've followed the instructions as outlined in the [PlatformIO Quick Start](https://github.com/Xinyuan-LilyGO/T-Display-S3?tab=readme-ov-file#3%EF%B8%8F%E2%83%A3-platformio-quick-start-recommended) from this repository.

At step 7 where it says "uncomment one of the lines" I tried using the tft line because I want to use the display.  This whole step is extremely vague and really needs to be rewritten, in my opinion.

I can build and upload that folder and it goes through the default Lilygo test pattern on the t-display s3 board.  This is the only thing I can get to work in PlatformIO.  I've had it working in the Arduino IDE but I'm trying to move everything to PlatformIO as it's supposedly better.

Now when I start my own project, I try to include TFT_eSPI it says it can't find it. No amount of copying files or editing platformio.ini files seems to work here.  If I add the TFT_eSPI by Bodmar library through the library manager and try using it, the board bootloops.

Can someone please explain how to set this up properly in PlatformIO because I'm at a loss here.  It's driving me crazy.

lewisxhe, 2024-10-01:
Using platform io does not require any setup, and there are many beginner tutorials on YouTube

teastainGit, 2024-10-01:
I don't know who told you PlatformIO was better. It is just Arduino IDE played on 'Hard" mode. The VAST majority of libraries and examples for microcontrollers are written for Arduino IDE in C, C++ as is Arduino IDE, using the built in Arduino libraries
 #include <arduino.h>.
Granted some special high performance applications that are written by professionals use functions that are only available in PlatformIO. Most of these are written in ESP-IDF which can be a world of pain to write, compile and upload even "Hello World".
I think ISP-IDF is a status symbol of superiority among hobbyist/makers. May of these Arduino critics have never even tried ver2, most of their complaints are about Arduino IDE from YEARS ago, lol!
All of the examples in this repository are written in Arduino IDE  and there is no improvement gained by compiling in PlatformIO! 
I use ver2.3.3 and am perfectly happy, but it is your choice.
-Terry

closed 2024-11-16 by github-actions[bot]