Xinyuan-LilyGO/T-Display-S3 #284: TFT display going OFF on LiPo battery power only

SimonePDA, 2024-11-07:
I have connected a PL 552035 3.7V 350mAh battery to the board connector. I am reading the voltage via software and it looks ok. When I disconnect the USB C connector, the unit keeps running but the TFT goes black. 
I also have two sensors connected to the QWIIC I2C 4 pin connector and they stay powered.

I am reading the battery voltage on GPIO 04, and when USBC is connected, I have a reading; the value goes to 0 when I run on the battery. 

This is my code:

```
#define BATT_PIN 4  // GPIO4 for battery monitoring
#define VOLTAGE_DIVIDER 2.0f
#define ADC_REFERENCE 3.3f
#define ADC_RESOLUTION 4095

float readBatteryVoltage() {
    // Read analog value
    int rawValue = analogRead(BATT_PIN);
    
    // Convert to voltage considering voltage divider
    float voltage = (rawValue * ADC_REFERENCE * VOLTAGE_DIVIDER) / ADC_RESOLUTION;
    
    return voltage;
}

```
Is there any specific PIN connection I need to make to keep the TFT ON?
Is the code wrong to read battery level when on battery power?

thanks

simone

lewisxhe, 2024-11-08:
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/2d3c2944c4fe5bdcc468d924667d80d947e8fc85/README.md?plain=1#L127

SimonePDA, 2024-11-08:
Perfect, the TFT going off is now fixed. 
The same code change suggested
```
      pinMode(15, OUTPUT);
      digitalWrite(15, HIGH);  //power the TFT backlight
```
fixed the battery issue as well: now I get the correct reading.

thank you

closed 2024-11-08 by SimonePDA