Xinyuan-LilyGO/T-Display-S3 #286: Problems with the display's output in Rainbow example code

tschimi1, 2024-11-13:
Hi, I am trying to get one of those to run. I installed the Rainbow example, installed the latest TFT-eSPI.h and had the whole stuff compiled and transfered into the ESP's memory. After a reboot, Screen comes up but somehow the size of the rainbow to be displayed does not fit the size of the display. I obviously followed all install procedures and also selected in the User_Setup_Select.h the required display.
Please also see the picture I have taken from the displays output in the attachment. 

Any help or hints would be much appreciated ;-)

Thx, tschimi1
[image]

lewisxhe, 2024-11-14:
The simplest troubleshooting is to copy the TFT_eSPI copy in the lib directory directly to the Arduino IDE library search path for testing. I can confirm that this example works properly.

tschimi1, 2024-11-20:
Hi, now I copied your library TFT_eSPI from the main branch. However, now I am getting errors related to GPIO.out_w1tc... Pls see my screenshot for further details. Many thanks in advance for having a look, tschimi1
[image]

lewisxhe, 2024-11-20:
Look here

https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/2d3c2944c4fe5bdcc468d924667d80d947e8fc85/README.md?plain=1#L64

[image]

tschimi1, 2024-11-20:
Sorry again. The version under your link is 2.5.0 for TTF_eSPI. Would you be so kind to point me to the 2.0.14 as this also does not seem to be available through Arduino?!
Thx, Torsten

tschimi1, 2024-11-20:
Finally got it working. Thx for all...

closed 2025-01-05 by github-actions[bot]