Xinyuan-LilyGO/T-Display-S3 #288: USB Host Support via GPIO pins or onboard connector

devilsbane98, 2024-11-20:
I am trying to connect a keyboard to the LilyGO T-Display-S3 AMOLED as an input device using USB Host mode. However, I am encountering the following issues:

1. The USB DM and USB DP pins, which are mentioned in the schematic, are not exposed anywhere on the board. This makes it unclear how to connect an external USB device directly.
2. The onboard USB-C connector does not recognize the keyboard when connected, despite attempting to configure the board for USB Host mode.

I have tried using:

- ESP-IDF's USB Host stack, and
- The ESPUsbHost library in Arduino,
but neither approach has worked for detecting or initializing the keyboard.

It seems there might be limitations in using the onboard USB-C connector or configuring the board for USB Host mode. Is there any official documentation or workaround for enabling USB Host functionality on this board? Additionally, is there a hardware solution (e.g., exposing DM/DP pins) to connect USB devices directly?

Any help or guidance on this matter would be greatly appreciated!

lewisxhe, 2024-11-21:
LilyGO T-Display-S3 AMOLED? Or LilyGO T-Display-S3? ?

devilsbane98, 2024-11-22:
I am using LilyGO T-Display-S3 AMOLED

closed 2025-01-06 by github-actions[bot]