Xinyuan-LilyGO/T-Display-S3 #291: Problems with battery charging

jperrela, 2024-12-02:
Hello,

I purchased some of your T-Display-S3 boards but have problems with the battery and charging.
Whenever I attach a battery and I charge it, after some circles, something breaks and the board stops working and charging. Then, if I try to power it on from the battery it gets stuck on a restarting loop. 
However, if I connect the board to USB (with or without the battery) works as expected, without chargin.

I am using this battery: https://tienda.bricogeek.com/baterias-lipo/135-bateria-lipo-1000mah-603050-37v.html
The board is encapsulated in resin. 

Could you please help us troubleshoot this? I can't find any documentation or troubleshooting about charging on your webpage.

Best,

lewisxhe, 2024-12-03:
Early esp32 core versions will cause the boot to get stuck when the USB is connected. If you turn on the esp from the battery, it will not run. You need to set USB ON CDC to off and then try again.

jperrela, 2024-12-05:
That did not solve the problem. I checked and R10 (a fuse) was slightly fused. What's the purpose of this fuse? Can I remove it if I'm using a battery that counts with overcurrent protections?

lewisxhe, 2024-12-06:
Has the fuse blown?
You need to measure whether the 3V3 voltage pad is short-circuited to GND. A blown fuse may mean that there may be a short circuit in the subsequent circuit.

jperrela, 2024-12-06:
It is not completely blown. It probably melted slightly since it is given 5ohms or so. 
I checked if 3,3V is short-circuited to GND and it is not.  Resistance values between 3.3V and GND are normal compared to a brand new board.

lewisxhe, 2024-12-07:
Have you measured the voltage behind the fuse after connecting the battery?

Is it the same for four boards in a row? What else did you connect besides the battery?

How did it become like this?

jperrela, 2024-12-07:
I have:

3,7V at battery pin

3V after R10

2,8V measured by the Esp32 using the software of the github examples folder. 

Voltages slightly vary for the other boards but the trend is the same.  How I reached this point is explained in the first message. 

Other things that are connected: a couple of sensors (pressure and distance sensor)

lewisxhe, 2024-12-07:
Have you removed all external sensors and then measured the voltage?

This board has only one voltage regulator, which is RT9013 from the schematic diagram. It can only provide 500mA current at most, which may not be enough for ESP32. The maximum peak current of normal ESP32 is generally around 700mA.

jperrela, 2024-12-07:
Sorry if my message was not clear: these measurements are without any sensor connected.

lewisxhe, 2024-12-07:
OK.
Now the information is clear.
Without any external sensors connected, only T-Display-s3
1. Measure 3.7V at the battery end
2. Measure 3V by skipping the fuse (R10)
3. Measure 2.8V through ADC

2.8V is not enough to start ESP32. Finally, you need to measure
1. How many V is 3v3 on the T-Display-s3 pin header when powered by battery.
2. How many V is 3v3 on the T-Display-s3 pin header after connecting to USB-C

jperrela, 2024-12-07:
With USB-C: 3.296V
With battery: 2.784V

lewisxhe, 2024-12-07:
If so, the only possible damage is Q4

I need to ask the hardware engineer on Monday about the location of Q4, and I need to short-circuit the two DS for testing.

[image]

jperrela, 2024-12-07:
Thanks a lot! I await for your answer.

BTW, in one of the units I short-circuited, the fuse and after many charging cycles, it seems to be working fine without sensors.  Though I'mdon't know the potential implications of this in the long term for the board or once I connect the sensors.

lewisxhe, 2024-12-09:
According to the picture, short-circuit these two points of Q4, do not use USB, use battery to connect the power supply, and check how many volts 3V3 is now
[image]

jperrela, 2024-12-09:
Done. It does not boot. No sensors are connected.

At battery pins: 3.82V
At 3V3: 2,74V

lewisxhe, 2024-12-10:
#include "soc/rtc_cntl_reg.h"
#include "soc/soc_ulp.h"

jperrela, 2024-12-10:
2.9V at the shorted point. The same voltage that I have after R10.

jperrela, 2024-12-10:
Battery is 3.8V.  R10 is around 45 ohm

lewisxhe, 2024-12-10:
Try short-circuiting R10 as well.

jperrela, 2024-12-10:
> Thanks a lot! I look forward to your answer.
> 
> BTW, in one of the units I short-circuited the fuse and after many charging cycles, it seems to be working fine without sensors. However, I don't know the potential implications of this in the long term for the board or once I connect the sensors.
> 
> So I don't know if there is anything damaged apart from the fuse, given that the voltage drop occurs just after it in the input voltage to the TP4065 from the battery (when the fuse is damaged and not short-circuited). So we have a minored input voltage and this could be the cause of having a minored voltage also in the 3V3. 

I already did without shortcircuiting Q4. Can you check this message?

lewisxhe, 2024-12-10:
I mean to connect R10 and Q4, and let the battery power directly connect to RT9013. It should work normally because it is normal when powered by USB.

jperrela, 2024-12-10:
To which point should I short circuit?

lewisxhe, 2024-12-10:
[image]
In this way, the battery voltage should be directly connected to the RT9013

jperrela, 2024-12-10:
When shortcircuiting both I get:
3.295V at 3.3V. But is there any difference in shortcircuiting Q4 or not? May this affect anything? I don't think Q4 is damaged.

lewisxhe, 2024-12-11:
Short-circuit Q4 and measure the 3.3v voltage. You have already done this above. So now you can remove the short-circuit point of Q4 and directly short-circuit R10 to test whether the 3.3V is normal. If it is normal, then R10 is damaged.

jperrela, 2024-12-11:
> That did not solve the problem. I checked and R10 (a fuse) was slightly fused. What's the purpose of this fuse? Can I remove it if I'm using a battery that counts with overcurrent protections?

That was clear from the beginning. But will it work normally and can I use the batteries safely without R10? My batteries have overcurrent protection. The link is on the conversation.

lewisxhe, 2024-12-11:
Of course, it can be used safely without R10. Generally, lithium-ion batteries have their own protection circuits.

closed 2025-01-26 by github-actions[bot]