Xinyuan-LilyGO/T-Display-S3 #293: Black screen on S3 T display

Deuf76, 2024-12-12:
Good morning, 

I have a S3 T display from lilygo; The screen was ok with some animation on it when I plugged it the 1st time.
Then, I have follow the instruction of the github by placing the folder in the librairies and also at the same levent the TFT folder. I have paid attention to not use recent esp32 board manager and adjusted all the different suggested configurations
I then try to use examples from the ide arduino or simple basic program to have something on the screen but nothing is appearing. :( 

I tried a simple program to switch on a led with the board to ensure that there was a good communication, it was fine.

How to move on ?
Thank you

[image]

example of basic programm for the display:

#include <TFT_eSPI.h> // Include the graphics library (needs to be installed)

TFT_eSPI tft = TFT_eSPI(); // Create an instance of the display

void setup() {
  tft.init(); // Initialize the display
  tft.setRotation(1); // Set the orientation of the display
}

void loop() {
  tft.fillScreen(TFT_BLACK); // Clear the screen to black
  tft.setTextColor(TFT_WHITE, TFT_BLACK); // Set the font color
  tft.drawString("Hello, T-Display-S3!", 10, 10, 2); // Draw a string on the display
  delay(2000); 
  }// Wait for 2 seconds

lewisxhe, 2024-12-13:
You ignored the precautions mentioned in the README

https://github.com/Xinyuan-LilyGO/T-Display-S3?tab=readme-ov-file#9%EF%B8%8F%E2%83%A3-faq

lewisxhe, 2024-12-13:
By the way, T-Display-S3-main should not be placed in the library directory, just copy the folder in lib to the library directory

Deuf76, 2024-12-13:
Hello, 
Thank you for your feedback.
I am not using battery but direct connection to the board from computer.
When I tried to launched the Arduino GFXD programm , I ve got this message (see picture)
Regarding the librairies, I've got the impression that I have follow the rules to place them

Thank you for your reply
Frederic

[image]
[image]

Deuf76, 2024-12-13:
Hello, 
Thank you for your feedback. I have followed your step but it is still black screen :(

Change library organisation:
[image]
Direct connection from the computer
Update TFt eSPI (bodmer 2;5.43)
in the file User_setup.select h I comment out the line //# include <User_Setup.h>           // Default setup is root library folder
and supress the 2 // in front of the line 206

everything is uploading well, but the screen is still black

what should be the next steps ?

Thanks again
Frederic

[image]
[image]
[image]
[image]

lewisxhe, 2024-12-16:
You are not using the TFT_eSPI example, but the GFX Library for Arduino. Please check whether the versions are the same. I tested it and it is normal.
[image]

closed 2025-01-31 by github-actions[bot]