Xinyuan-LilyGO/T-Display-S3 #297: factory example not displaying correct time

roblatour, 2025-01-13:
ardunio IDE 2.3.4

esp32 by espressif is set to 2.0.14

file - preferences - additional board managers - https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

file - preferences - sketchbook location set to a folder containing the factory example, with  
library files copied in a subfolder called 'libraries' of the factory project 

board selected as LilyGO T-Display S3
configured as recommend on github

in the pinconfig.h
   edited SSID and Password
   edited and uncommented 
         #define CUSTOM_TIMEZONE             "America/Toronto"

compile ok, 
debug session log shows:

wifi connected
timezone is correct

but 
display reads 12:34

lewisxhe, 2025-01-13:
I think the expiration date of the certificate has been clearly stated in the example, and it is obvious that the SSL certificate has expired.

roblatour, 2025-01-13:
yes, I looked at that but if you set your own custom time zone then the program does need to make the api call which requires that certificate.

roblatour, 2025-01-13:
after some more digging, its the following statement that is not doing what it should

in loop()

 if (getLocalTime(&timeinfo)) { ... }

is returning false (it should return true)

roblatour, 2025-01-13:
Looking at the code, I realized 

CUSTOM_TIMEZONE should be set to EST5EDT,M3.2.0,M11.1.0

Regardless, the it still fails:

in loop()

at

if (getLocalTime(&timeinfo)) { ... }

by returning false (instead of true)

lewisxhe, 2025-01-14:
Did you try my example of updating the certificate? I verified that it is normal. getLocalTime will only return if the time adjustment is successful, otherwise it will block there.

roblatour, 2025-01-14:
I'm pretty sure the certificate only comes into play when the CUSTOM_TIMEZONE is not set, in my case it is set to "America/Toronto".

In any case, I just now confirmed that while this code doesn't work when connecting to my regular network, even though other devices on it does, it does work when connected to my cell phone's hot spot.

Not sure why that is, but for now I'm going to close this issue.

With my sincere thanks for your help.

closed 2025-01-14 by roblatour

roblatour, 2025-01-14:
as a final update - I found that on my network if I assign a static ip address to the esp32 device then NTP works just fine.  In my testing above, when things were not working, the esp32 device had been assigned a dynamic address by my network.