Xinyuan-LilyGO/T-Display-S3 #300: Screen problems

sgbmzm, 2025-03-10:
Hello.
In general, I really enjoy this device.
I have a problem:
The screen turns white all around.
It doesn't stay fixed but after a few minutes it goes back to normal.
This probably happens more when I put the device in my pocket and it heats up - but not only.
Is this a known phenomenon?
And how do you solve it?
I'm attaching a picture. Don't pay attention to the yellow line, because I made it, just the white around it.
This happened on several different screens, not just one.
Thanks.

PS. Is its screen matte or glossy?
I'm asking because I once bought your original With Black Shell [K204], and the screen was glossy. And I also bought a Non-Soldered Pin [H569] on Aliexpress (Was it an imitation/cloning?) and the screen wa

[image]

s matte.

sgbmzm, 2025-03-10:
[image]

lewisxhe, 2025-03-10:
You said this problem exists in multiple devices, so I suspect it is caused by screen initialization. Have you tested it with sample code?

sgbmzm, 2025-03-10:
This doesn't happen on all devices and it doesn't always happen, but it happens on several different devices so it's not a specific device issue. On several devices running the same code - some have the problem right now and some don't. So I don't think it's a code issue.

I'm using the micropython from https://github.com/russhughes/s3lcd/tree/main

Here
https://github.com/russhughes/s3lcd/blob/main/firmware/GENERIC_S3_OCT_16M/firmware.bin

It also sometimes acts as if an old text is left on the screen.

Anyway, here's some sample code

```

import tft_config
import s3lcd

tft = tft_config.config(rotation=3) 
tft.init() 

main():
    tft.fill(0)
    tft.write(FontHeb25,f' {ABC DEF GHI}', 0, 50, s3lcd.GREEN, s3lcd.BLACK)
    tft.show()
    
    
    

LCD_POWER = Pin(15, Pin.OUT)
RD = Pin(9, Pin.OUT)
BACKLIGHT = PWM(Pin(38, Pin.OUT), freq=1000)

power_state = True

while True:
 
    if power_state:
         
        BACKLIGHT.duty(500)
        RD.value(1)
        LCD_POWER.value(1)

        main()
        
        time.sleep(1) 
        gc.collect()
        
    else:
        
        wake1 = Pin(14, Pin.IN, Pin.PULL_UP)
        
        esp32.wake_on_ext0(pin = wake1, level = esp32.WAKEUP_ALL_LOW)
        
        tft.fill(0) 
       
        BACKLIGHT.duty(0)
        RD.value(0)
        LCD_POWER.value(0)
        
        machine.deepsleep()

```

lewisxhe, 2025-03-10:
After the screen is initialized, refresh the screen to all black, and then turn on the backlight last.

sgbmzm, 2025-03-14:
It didn't help. But now I noticed something very important. **All the screen problems happen only after the device wakes up from a deep sleep of at least a few minutes, (with a battery connected to it during the entire deep sleep).**
If the device is not in deep sleep this does not happen even when the screen and power are turned off using
```
BACKLIGHT.duty(0)
RD.value(0)
LCD_POWER.value(0)
```

I conclude:
**The problems do not happen when the device is completely on (even if the screen is off), and do not happen when the device is completely off (disconnected from power)
The problems only happen when the device is in _deep-sleep_ mode for minutes, and then I wake it up.**

The problem is that I need deep sleep to save battery.
Without deep sleep the battery works for 8-10 hours and with deep sleep it works for several days.

How can I fix the problem with deep sleep? And is there another way to save battery?

Thank you!

lewisxhe, 2025-03-17:
Do you mean that when using battery, the screen will have problems after waking up from deep sleep?

If so, I can do a test.

sgbmzm, 2025-03-18:
Exactly. 
And what probably happens is that the last text that was on the screen before entering deep sleep remains on the screen even when the screen is off, and it is smeared on the screen during deep sleep. Maybe for some reason the screen receives electrical voltage during deep sleep even though I turn off the necessary pins as I did in the above code.
I think so because when I always turn on the screen, I see for a short time the last text that was on before entering deep sleep mode.
P.S. These problems do not occur on devices with the glossy screen, but only on devices with the matte screen.

lewisxhe, 2025-03-19:
I modified an existing simple test example, you can test it
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/examples/Sleep

closed 2025-05-08 by github-actions[bot]