Xinyuan-LilyGO/T-Display-S3 #301: Esp32 lilygo s3 battery powered

Zn0-w, 2025-03-20:
I cant seem to make it turn on via the battery pins. I have tried everything on github and reddit and the furthest i have gotten is that it turns on sometimes. But i had a extra one running on the factory software and it worked perfectly. The second i flash my own code with arduino ide it stops working with battery power. 

I have also included this in my code 
pinMode(15, OUTPUT);
   digitalWrite(15, HIGH);

I also have USB CDC On Boot disabled

I am new to coding and need a little help.

lewisxhe, 2025-03-20:
Provide your own minimal test code, or use our minimal test code

Zn0-w, 2025-03-20:
#include <TFT_eSPI.h> // Include the graphics library

TFT_eSPI tft = TFT_eSPI(); // Initialize the display

void setup() {
  tft.init();                      // Initialize the TFT display
  tft.setRotation(1);              // Set display orientation
  tft.fillScreen(TFT_BLACK);       // Fill the screen with black
  tft.setTextColor(TFT_WHITE);     // Set text color to white
  tft.drawString("Hello, World!", 10, 10); // Display "Hello, World!" at coordinates (10, 10)

pinMode(15, OUTPUT);
digitalWrite(15, HIGH);

}

void loop() {
  // Main loop remains empty
}

lewisxhe, 2025-03-22:
Well, you may not understand, you need to set GPIO to HIGH at the very beginning, there are so many examples, you should compare what is the difference

[image]

https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/examples/tft/tft.ino

closed 2025-05-05 by github-actions[bot]