Xinyuan-LilyGO/T-Display-S3 #302: GSM module RX and TX pins

dim3w, 2025-04-16:
Hi, i try IOT-GA6 GSM module to Lilygo T-Display-S3 Touch and i can't because the 16,21 pins and 17,18 pins are occuped by the touch screen:
I have tried 2,3 pins and i have not AT commands responding !
Can you help me please?

dim3w, 2025-04-16:
Thanks for your callback, i have tried SerialExample but with no success, AT commands not responding!
I would like to adapt this code to the t-display-S3 board: 
https://blog.uelectronics.com/tarjetas-desarrollo/arduino/como-cargar-los-comandos-at-al-modulo-iot-ga6/

PS: is it possible to use the display touch screen with IOT-GA6 GSM module

I have tried to update below code:

#include <Arduino.h>

const uint8_t rx1_pin = 2;
const uint8_t tx1_pin = 3;

void setup()
{
    Serial.begin(115200);
    Serial1.begin(9600, SERIAL_8N1, rx1_pin, tx1_pin);

    delay(5000);

    send_sms();
}

void loop()
{
    while (Serial1.available()) {
        Serial.write(Serial1.read());
    }

    while (Serial.available()) {
        Serial1.write(Serial.read());
    }

    delay(1);
}

void send_sms()
{
    Serial.println("Send sms...");

    Serial1.print("AT+CMGF=1\r"); 
    delay(100);
    Serial1.print("AT+CMGS=\"+52155-------\"\r"); 
    delay(200);
    Serial1.print("Hola, Saludos de CDMXELECTRONICA"); 
    Serial1.print("\r"); 
    delay(500);
    Serial1.print((char)26);
    delay(100);
    Serial1.println();

    Serial.println("SMS send");
    delay(500);
}

Thanks

lewisxhe, 2025-04-17:
You only set RX/TX, did you control your GSM modules? Did they power on? I only see the Serial loopback test in the code, I don't see you controlling the GSM power on and waiting for AT.

dim3w, 2025-04-20:
Hi, i tried another GSM module SIM800L and it works, AT commands are ok!
Thanks for your help

closed 2025-06-04 by github-actions[bot]