Xinyuan-LilyGO/T-Display-S3 #303: Midi Shield for T-display-S3 - Documentation

dim3w, 2025-04-20:
Hi i bought midi shield for t-display-s3, I'm looking for more documentation, PDF and video demonstration, do you have?

lewisxhe, 2025-04-21:
Schematics and examples are here.
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/examples/T-Display-S3-MIDI
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/schematic/SCH_T-Display-S3-MIDI_V1.1.pdf
What do you need ?

dim3w, 2025-04-22:
yes i saw it, but on the midi board, touch1 and touch2 can be press to produce musical notes ?

lewisxhe, 2025-04-23:
This requires you to control which button to press through the program and then make a sound!

dim3w, 2025-04-23:
there is a midi sound bank integrated?

lewisxhe, 2025-04-24:
No! It is just hardware. You can store audio files in the SD card for playback.

dim3w, 2025-05-02:
wich audio format it can plays?

closed 2025-06-16 by github-actions[bot]

my-project-droid, 2025-12-18:
what is the MIDI shield used for and where is there a earphone on the board