Xinyuan-LilyGO/T-Display-S3 #308: PLS, help I want to fix my T-display

fern-gpu, 2025-05-13:
My screen of the T-display is black, and it keeps giving me 
Sketch uses 378140 bytes (12%) of program storage space. Maximum is 3145728 bytes.
Global variables use 22180 bytes (6%) of dynamic memory, leaving 305500 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.8.1
Serial port COM5
Connecting....

A serial exception error occurred: Write timeout
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1 after I wiped the factory code (I did not read anything before hand)

lewisxhe, 2025-05-13:
[image]

fern-gpu, 2025-05-13:
Thank you though

lewisxhe, 2025-05-14:
Is the problem solved?

Still the same problem, you can use the simplest sketch to test whether it can run normally, such as da'ying

fern-gpu, 2025-05-14:
I have tried that before but it's still blank.

lewisxhe, 2025-05-15:
If all the above are invalid, please flash the factory firmware for quick verification, please check [here](https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/firmware/README.MD)

fern-gpu, 2025-05-18:
So, it still doesn't work (Error: Failed to execute 'open' on 'Serial Port': The port is already open.

lewisxhe, 2025-05-19:
Please make sure that the port is not used by other applications.

closed 2025-07-03 by github-actions[bot]