Xinyuan-LilyGO/T-Display-S3 #310: OV2640 camera fails after power cycle when using ESPHome

naamah75, 2025-06-06:
**Device**: LilyGO T-Camera S3
**Camera module**: OV2640
**Official working firmware**: firmware_t-cameras3_20230505.bin
**Custom firmware**: ESPHome

🔧 **Observed behavior**

- Flashing the official LilyGO firmware (firmware_t-cameras3_20230505.bin) initializes the OV2640 camera correctly.
- Immediately flashing a custom ESPHome firmware without disconnecting power allows the camera to continue functioning.
- However, after a power cycle (disconnecting and reconnecting power), the camera fails to initialize under ESPHome, producing the following log message:
- `[esp32_camera:187]: Got invalid frame from camera!`
- Reflashing the official firmware and then the ESPHome firmware (without a power cycle in between) restores camera functionality until the next power cycle.

❓ **Request for assistance**
I am seeking guidance on configuring ESPHome to ensure the OV2640 camera on the T-Camera S3 initializes correctly after a power cycle.

Specifically:

- Are there specific GPIO configurations or initialization sequences required for the camera module?
- Does the official firmware perform any hardware initializations that need to be replicated in ESPHome?
- Is there documentation or guidance available for integrating the OV2640 camera with ESPHome on the T-Camera S3?
- Any assistance or documentation you can provide to help resolve this issue would be greatly appreciated.

Thank you!

lewisxhe, 2025-06-07:
That's because you didn't configure the PMU to turn on the camera power.

Here is a related yaml configuration for reference
https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/master/esphome/t-camera-s3.yaml

naamah75, 2025-06-07:
yes, I saw that yaml file but unfortunately i couldn't use it because of the following error:

```
The "custom" component has been removed. Consider conversion to an external component.
https://esphome.io/guides/contributing#a-note-about-custom-components.
```

I am not able to write a custom component

lewisxhe, 2025-06-09:
I checked and it really doesn't work. Now I have fixed the compilation problem. You can try to see if it works. Because I don't have this board. I will close the issue first. If you still have problems, please go to t-camera-s3 to raise an issue. This repository is display-s3.

closed 2025-06-09 by lewisxhe